raspberrypi/pico-vscode #5: Compile projects fails under Windows

johnholman, 2024-03-29:
Installed in an existing VSC installation (1.87.2) under Windows 11.

Created a "new project", set the location and default options except for
board type: pico W
features: PIO interface
stdio support: console over UART
pico wireless options: background lwIP

The project is successfully generated and configured, but "compile project" fails with

* Executing task: C:\Users\johng/.pico-sdk/ninja/v1.11.1/ninja -C 'C:\Users\johng\Google Drive\src\picoproj\picotest/build'

bash: C:Usersjohng/.pico-sdk/ninja/v1.11.1/ninja: No such file or directory

* The terminal process "C:\Program Files\Git\bin\bash.exe '--login', '-i', '-c', 'C:\Users\johng/.pico-sdk/ninja/v1.11.1/ninja -C 'C:\Users\johng\Google Drive\src\picoproj\picotest/build''" terminated with exit code: 127.

I can see ninja has been installed at c:\Users\johng\.pico-sdk\ninja\vb1.11.1\ninja.exe but the bash command line looks wrong - missing slash separators between C: and Users and between Users and johng. As gsh says, looks like some backslashes need to be escaped.

paulober, 2024-03-29:
Thanks for reporting. I'll have a look.

will-v-pi, 2024-03-31:
It looks like the issue is that you're using  Git Bash on Windows, rather than the Command Prompt or PowerShell. Could you change the default profile to either of those (ctrl+shift+p then type & click Terminal: Select Default Profile), and see if that works for you?

We'll add git bash support on windows to the list of things to add support for, as that does seem useful to have

closed 2024-03-31 by paulober

paulober, 2024-03-31:
That's right. The issue was cause by the git bash.
By setting the type of the compilation task to 'process', the task is now independent of the standard shell profile.
Due to a bug in vscode https://github.com/microsoft/vscode/issues/209218 that resolves `${userHome}` incorrectly on Windows, I had to create a new platform-dependent command for Windows until the problem was fixed.

johnholman, 2024-04-01:
> It looks like the issue is that you're using Git Bash on Windows, rather than the Command Prompt or PowerShell. Could you change the default profile to either of those (ctrl+shift+p then type & click Terminal: Select Default Profile), and see if that works for you?

Yes, those both work.

paulober, 2024-04-01:
Does [this build](https://github.com/raspberrypi/pico-vscode/actions/runs/8499657306) also works with git bash as default profile on your PC?

johnholman, 2024-04-01:
How do I update to this?

paulober, 2024-04-01:
Download the artifact, unpack, uninstall the current version of the extension and select the three dots in the extension explorer (top right) and select install from vsix -> select the unpacked file.

johnholman, 2024-04-01:
Thanks - I just did this but still getting the same error message when using bash as the terminal shell.

paulober, 2024-04-01:
Sorry I forgot that it only applies to newly generated projects after you install the patch.

johnholman, 2024-04-01:
OK. Created a new project (same spec as before) and compilation now succeeds using bash.

(There does seem to be a problem with the generated code but I'll report that separately if it really is one).

Thanks!

commit 21a1709b (2024-03-31): Fix #5, Compile project fails on Windows
commit fd476eb4 (2024-03-31): Fix #5, Compile project fails on Windows