raspberrypi/pico-vscode #6: Error in generated code

johnholman, 2024-04-01:
Using prerelease 0.12.2 plus fix for issue #5 on windows 11.

 Created new project using defaults except for

name: picotest3
board type: pico w
location: ~/GD/src
features: pio interface
stdio: console over UART
pico wireless options: background LWIP

The generated code fails to compile with error in picotest3.c 
struct "_cyw43_t" has no field "netif"  (ln 54)

will-v-pi, 2024-04-03:
I cannot reproduce this - have tried with this configuration in my Google Drive local folder, and the code compiled without issue. There is a red squiggle under the netif, but that appears to be erroneous as it compiles correctly.

Can you try deleting the installed sdk folder (`C:\Users\johng\.pico-sdk\sdk\1.5.1`), and then deleting and recreating the project, to see if that compiles for you? This will install the SDK from scratch (in case that is the issue), and then re-generate the code.

johnholman, 2024-04-04:
Tried this and I think I may have been seeing the same as you did. That is, I still getting the same error about netif reported in "problems" panel but the code does compile and runs as expected (including reporting the IP address). So maybe this is a problem with intellisense in VSC (not an area I'm familiar with)?

will-v-pi, 2024-04-05:
Yeah, it looks like it is just an intellisense issue, I'll have a look at fixing that

closed 2024-04-05 by will-v-pi

paulober, 2024-04-05:
I also experienced this issue. It's caused by `CYW43_LWIP` not being set for intellisense. I tried implementing a change to set it in c_cpp_properties though this is only a workaround.

commit f0cf28e5 (2024-04-05): Generate compile_commands.json

Fixes #6, and possibly other intellisense bugs