raspberrypi/pico-vscode #14: Flash Task - openocd Error: Checksum mismatch - only the first time for every software change

simon-77, 2024-04-26:
## Steps to reproduce:
1. Setup a new _Raspberry Pi Pico Project_ in VSCode (e.g. _Blinky_ example project)
2. Programm the _Pico target microcontroller_ through a _Pico Debug Probe_ via the VSCode "Flash" task provided by VSCode Plugin.
3. When there is a change in the code (e.g. change the `sleep_ms(250)` value), there will be checksum mismatch error by openocd:

```
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
Warn : [rp2040.core1] target was in unknown state when halt was requested
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
** Programming Started **
Info : Found flash device 'win w25q16jv' (ID 0x001540ef)
Info : RP2040 B0 Flash Probe: 2097152 bytes @0x10000000, in 32 sectors

Warn : no flash bank found for address 0x00000000
** Programming Finished **
** Verify Started **
Error: checksum mismatch - attempting binary compare
** Verify Failed **
shutdown command invoked
```

This happens after code change, but then only for the first time the "Flash" task is called.
When the "Flash" task is called a second time, it will work reliably.

## Issue
It seems that the _Pico target microcontroller_ is not ready for flashing immediately, when some program was running before.

## Solution/Workaround\
Add a "delay" in openocd before programming.
Add a `after 100;` as argument to openocd for the Flash task in `.vscode/tasks.json`:
```
 {
            "label": "Flash",
            "type": "process",
            "command": "${userHome}/.pico-sdk/openocd/v0.12.0-2/bin/openocd.exe",
            "args": [
                "-f",
                "interface/cmsis-dap.cfg",
                "-f",
                "target/rp2040.cfg",
                "-c",
                "adapter speed 5000; after 100; program \"${command:raspberry-pi-pico.launchTargetPath}\" verify reset exit"
            ],
            "problemMatcher": []
        }
```

## Remarks about the required delay:
Some ad-hoc testing on a single machine resulted in the following:
- `after 40;` was reliably too little for a successful first-time upload 
- `after 50;` to `after 70;` worked sometime, failed sometimes for first-time upload
- `after 80;` did  reliably lead to a  successful first-time upload

Note:
I do not know if this is "the root cause" of the issue, or just a workaround for another issue. I am neither familiar with _openocd_ nor with the _Pico_ in particular.

Cheers,
Simon

will-v-pi, 2024-04-26:
It looks like this is actually a concurrency issue with the Compile and Flash tasks - can you confirm if you still see this checksum error if you run the Compile command before the Flash command, after changing the code?

simon-77, 2024-04-26:
You are right, if I do the compile separately, I does also work without the delay :+1: 

This was also my first thought, that's why I created a "Compile & Flash" task in VSCode, but apparently I have not configured that one correctly.

simon-77, 2024-04-26:
Although I have noticed that when calling the "Flash" task, also the ninja build is started.

So when doing a "Compile Project" task separately, and then running the "Flash" task, the ninja build did run 2 times ... :shrug: 

Maybe the "Flash" task is calling the ninja build and the openocd flash in parallel, instead of subsequently?

simon-77, 2024-04-26:
As workaround, I have created a "Compile and Flash" task, for which I have created a shortcut:

`tasks.json`:
```
...
,
        {
            "label": "Compile and Flash",
            "dependsOn": ["Compile Project", "Flash"],
            "dependsOrder": "sequence"
        }
```

`$USER/.config/Code/User/keybindings.json`:
```
// Place your key bindings in this file to override the defaults
[
    {
        "key": "ctrl+alt+f",
        "command": "workbench.action.tasks.runTask",
        "args": "Compile and Flash"
    }
]
```

will-v-pi, 2024-04-26:
Yes, that was the issue - it should be fixed with the referenced commit, so will be in the next release

will-v-pi, 2024-05-08:
Fixed in the latest release

closed 2024-05-08 by will-v-pi