raspberrypi/pico-vscode #17: Add a VSCode walkthrough

will-v-pi, 2024-05-20:
Add a walkthrough for users, to show the functionality of the extension (https://code.visualstudio.com/api/references/contribution-points#contributes.walkthroughs). This should include:
- Description of the extension and it's purpose
- Install instructions if on macOS (`xcode-select --install`)
- Which button to click for a new project/example
- The need to install the recommended extensions for debugging
- What settings are useful to change (eg features, correct debugger)
- How to connect the debugger (documentation links, or just diagrams)
- Which buttons to click for build & debug

kaycebasques, 2024-08-21:
Seems like you can close this now that the extension is covered in the main flow of [getting started with pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)?

neonmik, 2024-09-15:
There's still definitely more to be done here. Why is it downing python when I already have an up to date version? I know I need to specify the path, but that's not clear to a beginner.

neonmik, 2024-10-02:
> Because custom Python installations from users have caused too many issues in the past, so they can't be trusted anymore by default. If you want to use your system python, you can select "Use system version" under Python in the project generator after toggling "Advanced options" at the bottom.

That's fair, I hadn't noticed that option, it just felt unnecessary to download a whole different python... I get a bit sketch about having a load of hidden backend stuff, especially when I already have it!  

> Could you please elaborate a bit more on what specifically need further work?

Sorry, I was super vague here. There's some issues with project imports (specifically had issues with importing a large project seemed to be an issue with adding the new CMake to the original CMake file) , ended up starting a fresh and that helped.

I'm struggling to understand the use of compile options on the current extension vs the CMake build variants. I used to be able to choose between Debug/RelWithDebug/Release and now that whole tool bar seems to react badly with the new system. I feel like the compiler just optimised a load of stuff away that's broken my code in places - Probably more of a CMake issue though.

will-v-pi, 2024-10-03:
> I'm struggling to understand the use of compile options on the current extension vs the CMake build variants. I used to be able to choose between Debug/RelWithDebug/Release and now that whole tool [...]

By default the extension doesn't have an option to do this - you can either change it using the command line (run `cmake build -DCMAKE_BUILD_TYPE=Debug` from the integrated terminal), or use the CMake Tools extension integration which provides lots more CMake configurability (see https://github.com/raspberrypi/pico-vscode?tab=readme-ov-file#cmake-tools-extension-integration). I would recommend using the CMake Tools extension integration for this sort of thing.

neonmik, 2024-10-04:
> By default the extension doesn't have an option to do this - you can either change it using the command line (run `cmake build -DCMAKE_BUILD_TYPE=Debug` from the integrated terminal), or use the [...]

That's good to know, thank you. 

What optimisation does the plugin use as standard? I'm getting some wierd issues with it optimising out some important variables at the minute, even if they're marked Volatile.

will-v-pi, 2024-10-04:
> What optimisation does the plugin use as standard? I'm getting some wierd issues with it optimising out some important variables at the minute, even if they're marked Volatile.

It uses the default SDK build type, which is Release with debug info added.