raspberrypi/pico-vscode #25: Failed to setup example.

SeyranGrig, 2024-07-08:
Failed to setup example error when create new project from examples

SeyranGrig, 2024-07-08:
my version Beta 14 (RC 1)

will-v-pi, 2024-07-08:
Which example was this with? Additionally, what platform were you using (Windows, MacOS, Linux, etc)?

SeyranGrig, 2024-07-10:
I try on Windows any examples

SeyranGrig, 2024-07-10:
is it possible that the reason is that I have the Pico SDK v1.5.1 installed

SeyranGrig, 2024-07-10:
I remove installed Pico SDK v1.5.1, and i create project from example without errors

closed 2024-07-10 by SeyranGrig

Kabron287, 2024-07-18:
Windows10 x64
v0.14.0 (pre-release)
No separate SDK installation.

**Failed to setup example**

gokhan-bektas, 2024-07-27:
Same here.
Win 10 x64
VSCode 1.91.1
Raspberry Pi Pico Extension v0.14.0 (pre-release)

Tested with and without Windows Pico SDK installer installed.
Manually cloned examples to
 c:\users\myself\.pico-sdk\examples
and  to
 c:\users\myself\.pico-sdk\sdk\1.5.1\examples

Purged the VScode, all extension folders, all registry entries and re-installed. It is still the same.

Tried using 1.5.0, didn't make a difference.

Also tried to see if it is looking for a different folder:
[image]

will-v-pi, 2024-07-27:
Thanks, I've found the bug - there's some missing quotes, causing the space in the git path (Program Files\...) to cause a command failure. You can download the latest .vsix build from the actions artifacts [here](https://github.com/raspberrypi/pico-vscode/actions/runs/10124726049), and the fix will also be included in the next release

@gokhan-bektas As you've manually edited stuff in the .pico-sdk directory, I'd recommend deleting that whole directory to make sure everything is installed correctly, and then trying again with the updated extension

commit ca4c3176 (2024-07-27): Fix #25 - missing quotes around gitExecutable