raspberrypi/pico-vscode #27: Add options to include Pimoroni libraries

PJTewkesbury, 2024-07-13:
Hi, 

Loving the VSCode extension - Thanks
 
It would be great if you could add support for the Pimoroni hardware so all we need to do is select which hardware we have and the New Project wizard would add the correct config to get this compiling.

Thanks.

will-v-pi, 2024-07-15:
We can look into adding more boards to the drop-down selection - this would involve adding all boards supported by the SDK. For now, you can use different hardware by just changing the `set(PICO_BOARD ...)` line in the CMakeLists.txt file after the project is generated, and deleting the build directory and reopening the window - no other configuration should be needed.

PJTewkesbury, 2024-07-16:
Hi, thanks for your response. I was not thinking of adding more PICO boards to the list, but that would be good. I was thinking of adding options to the wizard to import libraries for the Pimoroni add on boards like the Pico_display_2 (https://shop.pimoroni.com/products/pico-display-pack-2-0?variant=39374122582099) and the others they produce.

Thanks.