raspberrypi/pico-vscode #30: c_cpp_properties has wrong forcedInclude path

IvanVeloz, 2024-08-09:
Using SDK 2.0.0, the Pico VS Code extension generates the following `c_cpp_properties.json`:

```
{
    "configurations": [
        {
            "name": "Pico",
            "includePath": [
                "${workspaceFolder}/**",
                "${userHome}/.pico-sdk/sdk/2.0.0/**"
            ],
            "forcedInclude": [
                "${userHome}/.pico-sdk/sdk/2.0.0/src/common/pico_base/include/pico.h",
                "${workspaceFolder}/build/generated/pico_base/pico/config_autogen.h"
            ],
            "defines": [],
            "compilerPath": "${userHome}/.pico-sdk/toolchain/13_2_Rel1/bin/arm-none-eabi-gcc",
            "compileCommands": "${workspaceFolder}/build/compile_commands.json",
            "cStandard": "c17",
            "cppStandard": "c++14",
            "intelliSenseMode": "linux-gcc-arm"
        }
    ],
    "version": 4
}
```

However, the path `${userHome}/.pico-sdk/sdk/2.0.0/src/common/pico_base/include/pico.h` doesn't exist. Although `${userHome}/.pico-sdk/sdk/2.0.0/src/common/pico_base_headers/include/pico.h` does exist. Changing that line fixes the highlighted syntax error.

This is on macOS, if that makes a difference.

paulober, 2024-08-12:
Thanks for reporting this issue. Will be fixed in the next release.

paulober, 2024-08-12:
For now you can find a preview build here: [Action run 10353453705](https://github.com/raspberrypi/pico-vscode/actions/runs/10353453705).

closed 2024-08-12 by paulober

IvanVeloz, 2024-08-12:
Thank you for the fast response!

commit ce3ae5ab (2024-08-12): Fix #30, c_cpp_properties sdk >=2.0.0 force include moved