raspberrypi/pico-vscode #31: picotool flagged as a virus

Nippius, 2024-08-09:
Hi,

Using the new extension on windows, when creating a new project for the very first time when the SDK, etc is downloaded and installed, the picotool that's downloaded is detected as a virus/trojan by windows defender:

Error message when I try to run the executable:
[image]

Event log generated:
[image]

For reference, I'm creating a new project that uses SDK 2.0.0 to targets the original RPi pico. The extension version is 0.15.1

Nippius, 2024-08-09:
Virus total has one detection so probably a false positive? https://www.virustotal.com/gui/file/31e55ed97a887fe59bbe8b3c200674c18a4626addc9aa2570871840cbbee3fa0/detection

will-v-pi, 2024-08-09:
Yes, definitely a false positive - what security intelligence version is your Windows Defender using? You can find this by going to Windows Security -> Virus & Threat protection and clicking on the Protection Updates link

Nippius, 2024-08-10:
Looks like the newest update fixes the false positive :)

Here's my security intelligence version:
[image]

Thanks for the help @will-v-pi

closed 2024-08-10 by Nippius