raspberrypi/pico-vscode #32: First time running new project from examples hangs on setup

mattmyne, 2024-08-10:
After installing the extension, the next step in the getting started guide is to create a "new project from examples".
As this is the first time, it needs to download the SDK etc. too.

For me, it then hangs showing the "Download and installing python. This may take a while..." and generating project notifications. While I appreciate it says this may take a while, I've left it over 20 minutes with a fast PC and internet connection and it has no activity after the initial button press.

The example project folder and files appear to be created ok.
The .pico-sdk folder is created and the examples sub-folder seems to be created and filled ok.
However, the python-->3.12.1 sub-folder is created but is empty and stays empty.

Task manager shows very low CPU and network usage for all processes - nothing appears to be happening.

I'm running Windows 11 (23H2) with the latest VS Code (1.92.1) and Pico extension (0.15.1).

I've tried deleting the .pico-sdk and project folders and trying again but the same thing happens each time. I've tried uninstalling and reinstalling the extension but again no change.

(Over a year ago I had manually installed the older SDK and associated tools but I had uninstalled all of this prior to my first attempt using this extension as I wanted a fresh start. I'd also removed the relevant environment variables. I'm reasonably sure there was nothing left behind that could be interfering with this new setup, but I'm adding this note just in case I've missed something).

PhilWatten, 2024-08-10:
I have a very similar issue on MacOS (14.5). In my case the python subfolder has two symbolic links to .exe files that point nowhere. On a mac, .exe isn't really a thing.

will-v-pi, 2024-08-10:
@mattmyne Could you try restarting your computer and creating a new project again? I don't see this issue on my Windows computers, but we have seen occasional download issues which can be fixed with a reboot. If that doesn't work then you can manually download the zip from https://www.python.org/ftp/python/3.12.1/python-3.12.1-embed-amd64.zip and unzip this into the .pico-sdk/python/3.12.1 folder, and then retry creating a project

@PhilWatten Those are the correct python symlinks - they're created so they are at the same path as on Windows. Can you elaborate on what the issue you saw was, as the Python installation on MacOS is done in a very different way to on Windows?

mattmyne, 2024-08-10:
@will-v-pi A reboot didn't fix it (I should have mentioned I'd also tried that). Thanks for the manual zip suggestion, hopefully the root cause can be found. I don't think there are any log files of use, but happy to provide any if there are.

mattmyne, 2024-08-10:
Following @will-v-pi 's suggestion I manually unzipped the python files into the empty folder and tried again. This time I got to the "Downloading SDK and Toolchain" message but it seems to have hung again. I see sdk and toolchain folders have been created and the sdk seems to have downloaded (git cloned?) ok, but I now have an empty ".pico-sdk\toolchain\13_2_Rel1" folder. As before, no further resource usage and it's been 20 minutes so I'm assuming it has stopped again.

mattmyne, 2024-08-12:
I guessed at the toolchain download and tried unzipping `https://developer.arm.com/-/media/Files/downloads/gnu/13.2.rel1/binrel/arm-gnu-toolchain-13.2.rel1-mingw-w64-i686-arm-none-eabi.zip` to the 13_2_Rel1 folder. It got a bit further (creating and filling some more folders) before getting stuck on I think the openocd files.

Before I unzipped the toolchain files manually, I could see the extension was downloading "something" fairly large before stopping. I saw no significant download activity after I manually unzipped the toolchain files and tried again. So I expect the extension is successfully downloading the file(s) but getting stuck unzipping or at least doing something with them afterwards.

There are no logs or other debug output that I can find and trying to debug purely on viewing resource usage and file explorer is pretty tenuous at best.

I tried downloading the extension source itself to debug in vscode but I don't normally develop in a typescript/npm environment and it's getting stuck at missing rollup commonjs packages (despite manually installing them), which I am unlikely to research further just to get the chance to further debug the extension.

will-v-pi, 2024-08-13:
To see the logs from the extension (and the rest of VS Code) you can click on Help -> Toggle Developer Tools, and then click Console. The library the extension uses for unzipping is slower than the system unzipping, so it's possible it's just taking a while and not really registering any CPU usage? It's also possible that your antivirus is doing something, as the files are all being downloaded from the internet.

I'm afraid I cannot reproduce this issue on any of my Windows machines, and the only times I've seen download problems on Windows they've been fixed by a reboot.

If you do want to try manual development, I have some build instructions at the bottom of the readme which should work - we use yarn for the dependency management

mattmyne, 2024-08-13:
Starting from scratch again, rebooted several times.
Only antivirus is built-in windows defender, I've disabled it.

Thanks for the developer tools console suggestion (though adding a more accessible log channel would be useful). However, not much is written and the last console message is `[Extension Host] [raspberry-pi-pico] Done copying example.` before the apparent hang while popup notification says downloading and installing python.

I can't imagine any unzip library is going to be as poorly performing as it would need to be to replicate what I'm seeing here. To be >30 minutes just unzipping a 7 mb openocd archive doesn't seem like a viable cause.

Instructions at the end of the readme had let me compile the .vsix using command line tools, but more work was needed to allow interactive debug of the code in the ide, which still isn't working as I'd expect (I can't find accessible breakpoints to trace what's happening, only managed to trigger in activate function of src/extension.mts so far).

Since you're unable to replicate this I'll have to assume it's something unique to my system. It's a fairly recent windows install with very little on it, but who knows! Thanks for looking into it.

jdbruner, 2024-08-15:
I'm seeing something similar on Ubuntu 24.04 (x86_64bit) - it hangs after downloading the example I've selected and the SDK, leaving me with an empty `.pico-sdk\toolchain\13_2_Rel1`

will-v-pi, 2024-08-15:
I've added a locations.md file in #42 with details of what the extension actually does, so if any downloads are failing then you can manually download and install the necessary files using that.

mattmyne, 2024-08-15:
I've been debugging the extension further. The issue (for my setup anyway) seems to be that the download is stopping part-way, with no apparent reason or error. For example, the 11 MB python zip reaches between 5 and 8 MB (varies each time I try) in a second or so, but then stops indefinitely.

It seems to be only an issue on the extension however. For example, running a simple node script:
```node
import { get } from "https";

const downloadUrl = 'https://www.python.org/ftp/python/3.12.1/python-3.12.1-embed-amd64.zip'

get(downloadUrl, response => {
  let chunks = 0;
  response.on("data", (chunk) => {
    chunks += chunk.length;
    console.log(`downloaded ${chunks} bytes`);
  });
  response.on("end", () => console.log(`finished downloading ${chunks} bytes`));
}).on("error", () => {
  console.log("download error");
});
```

...reports `finished downloading 11061655 bytes` as expected.

Adding similar `data` and `end` event logging to the `downloadEmbedPython` function in `download.mts` just hangs after displaying something like these last 5 output lines:

```
downloaded 6961941 bytes
downloaded 6978325 bytes
downloaded 6994709 bytes
downloaded 7011093 bytes
downloaded 7027477 bytes
```

It does this even if I replace most of the code in `downloadEmbedPython` with the simple code above - bypassing any request options, file writer, pipe functionality etc.

Unfortunately, at the moment I don't know why this is happening. If anyone knows why a vscode extension could be freezing part-way though a download when a normal node.js script doesn't, it'll be helpful!

mattmyne, 2024-08-17:
Replacing node's https get with the equivalent in the [undici](https://www.npmjs.com/package/undici) library works.
The undici library is popular and meant to have better performance compared with node https, but is this something we want to do?
If so I can replace all relevant code as a PR.

will-v-pi, 2024-08-19:
That sounds like a good solution then - if you could make a PR then that would be great, and then we can check it still works on all our systems

mattmyne, 2024-08-19:
This PR fixes the setup download stall issues on my system. To reduce the number of changes on the PR I've only updated src/utils/download.mts. There are further (minor) uses of https get in other utility files but they don't seem to directly affect this issue. Happy to update all files if deemed necessary as a separate issue/PR.

commit 918733bf (2024-08-23): Fix #32, Replace https get with undici client.stream in download.mts (#47)

Improves performance and fixes stalled downloads on some systems

---------
closed 2024-08-23 by will-v-pi