raspberrypi/pico-vscode #34: Users in China unable create project

CC52242, 2024-08-12:
I'm a developer in China. I can get this plug-in from vscode-market. **But I cannot creat project in this plug-in.**
1.  On the first try, I creat project in default setting, and an error pops up, it says, "Failed to download and install SDK and/or toolchain."
2.  After this, I tried to manually specify the PATH of GIT and other software, and also tried to modify the GITHUB token. But it didn't work.
3.  Then I tried setting up the http proxy and https proxy in Windows PATH, but it still didn't work.
I'm sure the proxy software I'm using is working properly. It could also be something I missed when setting up the proxy, but I have no clue.
I hope someone can provide a solution, such as an alternative mirror site that can be accessed by Chinese users, or allow the SDK's PATH to be manually specified in the plugin Settings. Or maybe, bring the windows standalone setup back? (By the way, this pack work's great on my pc. )

will-v-pi, 2024-08-13:
The extension will take quite a while to download and unzip everything on the first run - it will only create the project folder once everything has been downloaded.

Do you get any errors when trying to create a project using the proxy?

CC52242, 2024-08-13:
> The extension will take quite a while to download and unzip everything on the first run - it will only create the project folder once everything has been downloaded.
> 
> Do you get any errors when trying to create a project using the proxy?

From the first time I tried to download without an proxy to the time I adjusted (probably) all the adjustable Settings,
It just says "Failed to download and install SDK and/or toolchain."
It can load the project configuration interface with or without using the proxy. But it cannot go futher. Seconds after clicking the Create button, the error message popped up.

will-v-pi, 2024-08-15:
I'm afraid that without being in China I don't think I'm going to be able to debug this. I've created a locations.md file in #42 which has details for manually installing all the required tools - I would suggest you give that a go, if the extension is failing to download them?

We'd also welcome any PRs from other users in China if they have any solutions

cuicuixuan, 2024-10-09:
well i'm also a developer in China ,i met that problem too and annoyed me for atleast two weeks ,now i fixed that problem i mean your proxy was not configured correctly,In the case of clash verge, you need to run in tun mode， tun mode is a virtual network device that intercepts all network traffic from the operating system and redirects it to clash for proxiing, so that you can download the required toolchain normally hope that reply can help to you

umeiko, 2024-11-20:
My problem is when installing toolchain, shown faild and then stucked. I'm also in China mainland

will-v-pi, 2024-12-17:
See the comment on #136 - if you change your language settings to English (US) it seems to fix that issue

IotaHydrae, 2024-12-21:
Well, I have another problem.

> [!CAUTION]
> ERR [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - githubREST] Fetching pico-sdk-tools releases failed: GitHub API Code 403 Forbidden. Rate limit exceeded.

So I add a personal token and I got a new one.

> [!CAUTION]
> ERR [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - githubREST] Fetching pico-sdk-tools releases failed: Error http status code: 404

I tried it several times, but the result was the same.