raspberrypi/pico-vscode #37: Using PROJECT_NAME variable in cmake file does not work

CHiPs44, 2024-08-14:
When I `set(PROJECT_NAME foobar)` in `CMakeLists.txt` file and use it as `project(${PROJECT_NAME} C CXX ASM)`, I get this kind of errors when trying to run the project:

```
 *  Executing task in folder foobar: /home/user/pico-sdk/picotool/2.0.0/picotool/picotool load /home/user/src/foobar/build/${PROJECT_NAME}.elf -fx 

ERROR: Could not open '/home/user/src/foobar/build/${PROJECT_NAME}.elf'
```

If I replace every occurrence of `${PROJECT_NAME}` with `foobar`, everything runs fine and UF2.

NB: compiling project works with `PROJECT_NAME` variable.

will-v-pi, 2024-08-14:
Yes, this won't work as the extension just parses the text from that line. By default this extension only works with CMake projects containing a single executable, where the executable name matches the project name, and the project name is not a variable in the CMakeLists.txt file.

If you want to use a variable for the project name, then you'll need to tell this extension to use the CMakeTools extension, which should work with any CMakeLists.txt file. You can do this by modifying the follow settings in the .vscode/settings.json file
```
"raspberry-pi-pico.cmakeAutoConfigure": false,
"raspberry-pi-pico.useCmakeTools": true,
```
You can optionally also set these to true
```
"cmake.configureOnEdit": true,
"cmake.automaticReconfigure": true,
"cmake.configureOnOpen": true,
```
This will cause this extension to use the CMakeTools extension for compilation, so you may need to click the CMake icon on the left to set it up correctly. If prompted to select a kit, choose the Pico kit.

CHiPs44, 2024-08-14:
Thanks for the FTL explanation!

closed 2024-08-14 by CHiPs44