raspberrypi/pico-vscode #38: Error "Cannot read properties of undefined (reading 'replaceAll')" after selecting SDK 2.0.0

CHiPs44, 2024-08-14:
After selecting SDK 2.0.0 (from 1.5.1), I get a notification "Cannot read properties of undefined (reading 'replaceAll')" Developer tools console says:

[image]

paulober, 2024-08-16:
Thanks for reporting. Can you confirm if the issue still exists on the current [nightly release](https://github.com/raspberrypi/pico-vscode/actions/runs/104066161580)?

CHiPs44, 2024-09-05:
Oops! I missed the notification, will try this evening at home.

paulober, 2024-09-17:
@CHiPs44 Any results?

CHiPs44, 2024-09-20:
Same message with [0.16.2] - 2024-09-17.

will-v-pi, 2025-02-25:
Are you still having this issue with the latest extension version? And if you are, is it affecting the functionality of the extension, or is it just an error in the developer console but the project creation is still working fine?

I can't reproduce this myself with the latest version, so I believe this is fixed

closed 2025-04-15 by will-v-pi