raspberrypi/pico-vscode #40: FYI: missing libs for openocd on Ubuntu 22.04: libftdi1-2 libhidapi-hidraw0

CHiPs44, 2024-08-15:
I had to install `libftdi1-2` and `libhidapi-hidraw0` to be able to use openocd with my picoprobe:

```bash
sudo apt install libftdi1-2 libhidapi-hidraw0
```

kaycebasques, 2024-08-21:
I think I also just ran into this while trying the [debug](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf#page=11) workflow from Pico quickstart

```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from objdump-multiarch --syms -C -h -w /home/kayce/pico/projects/blink/build/blink.elf
Reading symbols from nm-multiarch --defined-only -S -l -C -p /home/kayce/pico/projects/blink/build/blink.elf
Launching GDB: gdb-multiarch -q --interpreter=mi2
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Error: nm-multiarch failed! statics/global/functions may not be properly classified: Error: spawn nm-multiarch ENOENT
    Expecting `nm` next to `objdump`. If that is not the problem please report this.
Error: objdump failed! statics/globals/functions may not be properly classified: Error: spawn objdump-multiarch ENOENT
    ENOENT means program not found. If that is not the issue, please report this problem.
Launching gdb-server: /home/kayce/.pico-sdk/openocd/0.12.0+dev/openocd.exe -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/kayce/.pico-sdk/openocd/0.12.0+dev/scripts -f /home/kayce/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000"
    Please check TERMINAL tab (gdb-server) for output from /home/kayce/.pico-sdk/openocd/0.12.0+dev/openocd.exe
Finished reading symbols from objdump: Time: 13 ms
Finished reading symbols from nm: Time: 10 ms
```

After installing the stuff that @CHiPs44 mentions it seems to be working now

will-v-pi, 2024-08-27:
I've added a note on this to the readme

commit c845f470 (2024-08-27): Fix #40, add missing openocd libs on Ubuntu 22.04 to readme