raspberrypi/pico-vscode #46: Improve project importing and CMake Tools integration for pico-examples

will-v-pi, 2024-08-19:
The pico-examples repository is just one example, but there are a couple of improvements that I think will help with importing other projects too:

- Add/replace `set(PICO_BOARD xxx CACHE STRING "Board type")` to the imported project CMakeLists.txt - and also add board selection to the Import Project webview
- Detect whether a project will require CMake Tools, and switch to that (maybe with a popup to ask the user) - this will be for any project where there are multiple executables (eg pico-examples)
- As part of this, maybe add a command to toggle between CMake Tools and this extension for compiling - this would toggle
  - `"raspberry-pi-pico.cmakeAutoConfigure": true` -> `false`
  - `"raspberry-pi-pico.useCmakeTools": false` -> `true`
  - `"cmake.configureOnEdit": false` -> `true`
  - `"cmake.automaticReconfigure": false` -> `true`
  - `"cmake.configureOnOpen": false` -> `true`
- Use the "Configure CMake" button to trigger configuring the CMake Tools extension (like we do for the launchTargetPath stuff)
- Remove "toolchainFile" from generated `cmake-kits.json` - it's not necessary, as the SDK configures the correct one

jdbruner, 2024-12-03:
Another suggestion: When importing an existing project, check if its ```CMakeLists.txt``` already includes the VSCode extension block (marked by the "DO NOT EDIT THE FOLLOWING LINES for the Raspberry Pi Pico VS Code Extension to work" comments). If so, replace it with the newly-constructed block. Currently, another block is prepended to the file (requiring manual removal of the duplicate).

BTW, should the ```set(PICO_BOARD xxx CACHE STRING "Board type")``` line be inside this block instead of immediately following it?

will-v-pi, 2024-12-03:
You shouldn't need to import an existing project - if it already has those lines at the top it should work with the extension without needing an import. If you have a project like that which does require an import to work, could you share it so we can figure out why it's not being detected as a pico project?

The `PICO_BOARD` line is outside that block because it can be modified by a user, and some imported projects may wish to have it in a different place in the file, so keeping it outside the block means that the structure of the block is the same for all projects.

trautb, 2025-01-16:
> You shouldn't need to import an existing project - if it already has those lines at the top it should work with the extension without needing an import. If you have a project like that which does [...]

I'm working on a project that I think somehow fits here.
Everytime I create a new branch/worktree and open it in VSCode the extension recognizes the project (e.g. `PicoSDK version` and `board` shown in the lower right corner, see image).

[image]

But `Configure CMake` yields the error
```Development for the Pico requires Ninja to be installed and available in PATH. Please install and restart VS Code.)```
I usually use the version of the build toolchain automatically downloaded by the extension and would assume that either a discovery of the already downloaded versions or an additional but automatic download of the toolchain would solve this.

[image]

will-v-pi, 2025-01-20:
That sounds like a different issue - try using the latest debug build of the extension to see if you still have the same issue, and if you're still having problems then create a new issue including full details of your project.