raspberrypi/pico-vscode #49: Only show Pico icon in activity bar when working on Pico projects

kaycebasques, 2024-08-21:
Just installed the extension. It seems that the Pico icon is now always present in my VS Code, even when I'm not working on Pico projects. Can you update the extension to only show the Pico icon in the [activity bar](https://code.visualstudio.com/api/ux-guidelines/activity-bar) when I'm actually working on Pico-related projects?

will-v-pi, 2024-08-22:
That icon also gives access to the project creation pages, so it still needs to be displayed when you're not working on Pico projects, to allow creation of new Pico projects. If you want to hide the icon when not working on Pico projects then you can right-click and hide it, or look into using Profiles as suggested in #43

kaycebasques, 2024-08-22:
I guess it depends on how frequently you expect users at large to spin up new Pico projects and how much of their VS Code usage is dedicated to Pico. I work on projects in many different domains. The chances that I'm going to want to spontaneously spin up a new Pico project when working on one of my websites, ML projects, Python libs, etc. is low. Creating new projects is also not that frequent of an action (compared to exploring files for example) so I'm not sure that it warrants having an always-present icon in a place as prominent as the activity bar.

Perhaps consider hiding it when a Pico project is not detected and providing a command (in the command palette) to re-show the icon for when the occasional task of creating a new project comes up?

Anywho, love Pico and love the extension. Just figured I'd put this on your radar. Feel free to close this issue

will-v-pi, 2024-08-25:
We've added a note about VS Code profiles to the [readme](https://github.com/raspberrypi/pico-vscode#vs-code-profiles), which I think is a better solution than uninstalling and reinstalling - you can then switch to the profile when you want to create a new project, or edit an existing one

closed 2024-08-25 by will-v-pi