raspberrypi/pico-vscode #50: Reopening project sometimes requires "Import Project"

wiredopposite, 2024-08-21:
Sometimes opening an existing pico-vscode project requires importing it again with the extension, otherwise the configure and build buttons are greyed out. This also adds a bunch of unnecessary lines to your CMakeLists file and downloads the sdk and toolchain files over again.

will-v-pi, 2024-08-23:
This can occur if you don't have a `set(PICO_BOARD ....)` line in your CMakeLists.txt - is that the case here?

Could you also add your CMakeLists.txt file to this issue so we can see what else might be causing the problem

will-v-pi, 2024-09-25:
Closing this due to no response - let us know if you're still having this issue with the latest version of the extension

closed 2024-09-25 by will-v-pi

tomtom7899, 2024-09-26:
Hey, I have the same issue, here is my CMakeLists.txt

```
# Generated Cmake Pico project file

cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)

# == DO NEVER EDIT THE NEXT LINES for Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.0.0)
set(toolchainVersion 13_2_Rel1)
set(picotoolVersion 2.0.0)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================
set(PICO_BOARD pico CACHE STRING "Board type")

# Pull in Raspberry Pi Pico SDK (must be before project)
include(pico_sdk_import.cmake)

project(main-fw C CXX ASM)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

# Add executable. Default name is the project name, version 0.1

add_executable(main-fw main.c )

pico_set_program_name(main-fw "main-fw")
pico_set_program_version(main-fw "0.1")

# Modify the below lines to enable/disable output over UART/USB
pico_enable_stdio_uart(main-fw 0)
pico_enable_stdio_usb(main-fw 1)

# Add the standard library to the build
target_link_libraries(main-fw
        pico_stdlib)

# Add the standard include files to the build
target_include_directories(main-fw PRIVATE
  ${CMAKE_CURRENT_LIST_DIR}
  ${CMAKE_CURRENT_LIST_DIR}/.. # for our common lwipopts or any other standard includes, if required
)

# Add any user requested libraries
target_link_libraries(main-fw 
        hardware_spi
        hardware_dma
        hardware_clocks
        )

pico_add_extra_outputs(main-fw)
```

It is completly autogenerated and I didnt change anything

will-v-pi, 2024-09-26:
I'm afraid I cannot reproduce this issue with that CMakeLists.txt file - when I create an auto-generated project which creates that exact CMakeLists.txt file, then close and re-open it, the extension loads correctly and doesn't require importing the project.

What platform are you using the extension on (Windows, Raspberry Pi OS, etc)? Could you also try the latest build which you can download from the [actions](https://github.com/raspberrypi/pico-vscode/actions) tab and see if you have the same issues?

tomtom7899, 2024-09-26:
Im on Debian 12. Manually  importing the project doesnt help either so I think its not exactly the same issue. If you want, I can open a new issue.

I will try a newer version and i will also try to generate a new project and copy all the files over to see if the issue persists.

tomtom7899, 2024-10-02:
The problem still persists with the version from the latest master commit