raspberrypi/pico-vscode #51: Uninstall from VSCode fails

henrygab, 2024-08-23:
This extension refuses to uninstall.

Repro:
1. Install the extension.
2. Attempt to uninstall the extension by pressing the `Uninstall` button
[image]

Actual results:
Pressing the uninstall button appears to do nothing.

Expected results:
The extension should uninstall cleanly.

will-v-pi, 2024-08-24:
The uninstall is handled entirely by VS Code, not by this extension, so I don't think this is an issue we can solve. You could try restarting VS Code and see if it works then, and also try uninstalling other extensions to see if this is specifically an issue with this extension. If all else fails then you can delete the extension from `~/.vscode/extensions`

will-v-pi, 2024-08-27:
I think this has been fixed with that PR - could you download the .vsix file from the actions run (https://github.com/raspberrypi/pico-vscode/actions/runs/10565856890), install that, and then try uninstalling it and it should work cleanly?

henrygab, 2024-08-27:
Glad to hear you discovered a root cause, rather than:

> The uninstall is handled entirely by VS Code, not by this extension, so I don't think this is an issue we can solve.

That said, I've lost my appetite for this extension.  After all, I had to uninstall it because it was not a smooth experience with existing CMake projects.

PR #59 (2024-08-30): Use extension dependencies
Use extension dependencies, rather than installing as an extension pack

This fixes uninstall issues, where uninstalling the extension would also uninstall the Microsoft C/C++ extensions

This now requires a separate package for OpenVSX, as that package can't depend on the Microsoft C/C++ extensions, as they're not availble on OpenVSX

Fixes #51 and #55
closed 2024-08-27 by henrygab