raspberrypi/pico-vscode #52: Can't upload directly from VSCode

detoledovictor, 2024-08-24:
When play RUN button, after compling a project the terminal shows this message:

 Executing task: C:\Users\viict/.pico-sdk/picotool/2.0.0/picotool/picotool.exe load d:/Perfil/Documents/UNISAL/Arquitetura e Aplicacao/blink/build/blink.elf -fx 

ERROR: The RP2040 device returned an error: <unknown>

 *  The terminal process "C:\Users\viict\.pico-sdk\picotool\2.0.0\picotool\picotool.exe 'load', 'd:/Perfil/Documents/UNISAL/Arquitetura e Aplicacao/blink/build/blink.elf', '-fx'" terminated with exit code: -99.

will-v-pi, 2024-08-24:
Could you double check your RP2040 is in bootsel mode, that there is no debugger connected (and you've unplugged the board since the debugging finished), and that you have a driver installed using Zadig (see the picotool section of the Getting Started Guide for more details)? This is a picotool error, rather than an issue with the extension, so I'll transfer it there if you're still having problems.

will-v-pi, 2024-09-25:
Closing this due to no response - if you're still having this issue then let us know

closed 2024-09-25 by will-v-pi