raspberrypi/pico-vscode #53: Flashing does not work inside of workspaces

ambiguousname, 2024-08-24:
The `Flash` command is reliant on the command `${command:raspberry-pi-pico.getTarget}`, which tries to locate `CMakeCache.txt` before failing. Because I'm using a workspace, CMakeCache.txt is in a subdirectory.

For now, my workaround fix is just to set the target manually, but it'd be nice to have some sort of configuration option so I can point to where CMakeCache.txt is.

So far this command and the Quick Access panel seem to suffer from this issue, and everything else works fine inside of a workspace with some minor tweaking.

will-v-pi, 2024-08-26:
The extension doesn't support this by default - we only support having a single project open per window, with the CMakeLists.txt file in the root. You can switch the extension to using the CMake Tools extension which should be able to fix this - see #37 for more details, and #46 for how we're thinking of integrating this into the import project wizard

ambiguousname, 2024-08-30:
I see, thank you!

closed 2024-08-30 by ambiguousname