raspberrypi/pico-vscode #54: Failed generating project

Bazmundi, 2024-08-25:
Generating project based on the blinky example in d:\OneDrive\Documents\GitHub\blink-2040, this may take a while...: Failed

It also fails on import in other circustances and especially fails to download sdk?

will-v-pi, 2024-08-26:
Can you post the developer log you get when it fails - this can be found at Help -> Toggle Developer Tools -> Console? Also, is this with version 0.15.2, as we improved the downloading with that latest version?

stelladraco27, 2024-08-29:
I am unable to install this example, here is my log.
My version is 0.15.2
Here is the log from the developer tools once I click "new project from example", input "blinky" into name, input "pico" into board type, input project folder path, and the click on "Create":

```console
console.ts:137 [Extension Host] [raspberry-pi-pico] Using cached response for pico-sdk releases
console.ts:137 [Extension Host] [raspberry-pi-pico] Successfully downloaded examples list from the internet.
console.ts:137 [Extension Host] [INFO] [NewProjectPanel] Loaded 84 examples.

tailwindcss-3_3_5.js:62 cdn.tailwindcss.com should not be used in production. To use Tailwind CSS in production, install it as a PostCSS plugin or use the Tailwind CLI: https://tailwindcss.com/docs/installation
(anonymous) @ tailwindcss-3_3_5.js:62
(anonymous) @ tailwindcss-3_3_5.js:62

main.js:427 set theme dark

console.ts:137 [Extension Host] rejected promise not handled within 1 second: Error: Command failed: cd "C:/Users/Me/.pico-sdk/examples" && "P:\Path\To\Git\cmd\git.EXE" rev-parse HEAD
fatal: not a git repository (or any of the parent directories): .git
y @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
S @ rpcProtocol.ts:458
Q @ rpcProtocol.ts:443
M @ rpcProtocol.ts:373
L @ rpcProtocol.ts:299
(anonymous) @ rpcProtocol.ts:161
B @ event.ts:1230
fire @ event.ts:1261
fire @ ipc.net.ts:652
K.onmessage @ localProcessExtensionHost.ts:378

console.ts:137 [Extension Host] stack trace: Error: Command failed: cd "C:/Users/Me/.pico-sdk/examples" && "P:\Path\To\Git\cmd\git.EXE" rev-parse HEAD
fatal: not a git repository (or any of the parent directories): .git
    at genericNodeError (node:internal/errors:984:15)
    at wrappedFn (node:internal/errors:538:14)
    at ChildProcess.exithandler (node:child_process:423:12)
    at ChildProcess.emit (node:events:519:28)
    at maybeClose (node:internal/child_process:1105:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:305:5)
y @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
S @ rpcProtocol.ts:458
Q @ rpcProtocol.ts:443
M @ rpcProtocol.ts:373
L @ rpcProtocol.ts:299
(anonymous) @ rpcProtocol.ts:161
B @ event.ts:1230
fire @ event.ts:1261
fire @ ipc.net.ts:652
K.onmessage @ localProcessExtensionHost.ts:378

log.ts:439 ERR Command failed: cd "C:/Users/Me/.pico-sdk/examples" && "P:\Path\To\Git\cmd\git.EXE" rev-parse HEAD
fatal: not a git repository (or any of the parent directories): .git
: Error: Command failed: cd "C:/Users/Me/.pico-sdk/examples" && "P:\Path\To\Git\cmd\git.EXE" rev-parse HEAD
fatal: not a git repository (or any of the parent directories): .git
    at genericNodeError (node:internal/errors:984:15)
    at wrappedFn (node:internal/errors:538:14)
    at ChildProcess.exithandler (node:child_process:423:12)
    at ChildProcess.emit (node:events:519:28)
    at maybeClose (node:internal/child_process:1105:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:305:5)
```

Bazmundi, 2024-08-30:
> Help -> Toggle Developer Tools -> Console

Not as such, I used a manual install method suggested by https://github.com/grblHAL/RP2040/discussions/82.

I had nothing to do with the examples, as I tried installing grblHAL for RP2040 first.  Got the message. Ripped out the kit.  Reinstalled and then tried the examples.  I was using 0.15.2.

As I have the build up and running, using the guide above, I am dropping this.

Bazmundi, 2024-08-30:
> Can you post the developer log you get when it fails - this can be found at Help -> Toggle Developer Tools -> Console? Also, is this with version 0.15.2, as we improved the downloading with that latest version?

Take care in you're writing here.  Are you saying there is another version later than 0.15.2 that "fixes" a problem?  I have auto update set and 0.15.2 is the latest, or I assume its the latest, since it is Code is not prompting for a later install.  Hence, you may not have improved the downloading, since the problem is reported against 1.15.2.

will-v-pi, 2024-09-08:
@stelladraco27 I think your issue may be related to different drives (C: vs D:), so may be fixed by #74

will-v-pi, 2024-09-11:
I think this issue has been fixed in 0.16.0, so closing this issue

closed 2024-09-11 by will-v-pi