raspberrypi/pico-vscode #55: extension refuses to uninstall

Bazmundi, 2024-08-25:
Good job.  I thought I woudl uninstall then reinstall the extension to try to "fix" the problems I reported at https://github.com/raspberrypi/pico-vscode/issues/54 

If I try to disable the extension I get:

```
Cannot disable 'C/C++' extension alone.  'Blender Development' and 'PlatformIO IDE' extensions depend on this. Do you want to disable all these extensions?
```

I am not having to rip out MSCode all together to sort this nuasance out am I?

will-v-pi, 2024-08-26:
It looks like this is related to our use of an Extension Pack, which therefore uninstalls the entire pack when you click uninstall. I'm able to reproduce #51 now, if I install both this extension and the PlatformIO IDE before attempting to uninstall this extension.

We can investigate using dependencies instead, which might solve this issue. Although, uninstalling and reinstalling this extension is very unlikely to be a useful debugging step anyway, due to the nature of how VS Code extensions work.

will-v-pi, 2024-08-27:
I think this has been fixed with that PR - could you download the .vsix file from the actions run (https://github.com/raspberrypi/pico-vscode/actions/runs/10565856890), install that, and then try uninstalling it and it should work cleanly?

stelladraco27, 2024-08-29:
I tried installing the .vsix file. It does not help with uninstallation. I also tried reverting to an older version, didn't work either.

will-v-pi, 2024-08-29:
After you installed that .vsix file, did it still have an Extension Pack tab on the marketplace page, or did it have Dependencies tab instead? Also, could you elaborate on what errors you see when uninstalling, and possibly send a screenshot, and a list of other Extensions you have installed?

closed 2024-08-30 by will-v-pi