raspberrypi/pico-vscode #63: Import Project

Octopus1633, 2024-08-31:
Hello, 

What are the requirements of pico-vscode for imported projects? I can successfully import pico-examples and compile, but when I import my own project, I cannot import it successfully without any prompt.

I noticed that there is such a prompt on the import project interface:
Basic Settings
Warning: Project Import Wizard may not work for all projects, and will often require manual correction after the import

paulober, 2024-09-02:
Hi,
do you have any output in the "Output > Extension Host" panel?
and do you have a small example project on GitHub that silently fails when importing?

jancumps, 2024-09-02:
I have similar experience: New Project From Example works end-to-end.

New Project and Import Project fail silently, after flashing a brief Activating Extensions... notification on the bottom bar.
"Output > Extension Host" is empty. No message logged by the extension.

visual studio 1.19.2
plugin raspberry-pi.raspberry-pi-pico [0.15.2] - 2024-08-23

will-v-pi, 2024-09-04:
> I have same problem. Will not import this project: https://github.com/ThorstenBr/A2DVI-Firmware

I don't see a CMakeLists.txt file in the root of that repository - this extension only works with CMake projects, so requires a CMakeLists.txt file in the root of the project directory

paulober, 2024-09-05:
> I have similar experience: New Project From Example works end-to-end.
> 
> New Project and Import Project fail silently, after flashing a brief Activating Extensions... notification on the bottom bar. "Output > Extension Host" is empty. No message logged by the extension.

Can you try the latest [debug build](https://github.com/raspberrypi/pico-vscode/actions/runs/10717465754) (so the vsix file prefixed with debug-) and see if you get any output in the developer console (`Help > Toggle Developer Tools`) when opening a new vscode window with no open folder and then opening our sidebar panel or just running the new project command via the command palette `Ctrl+Shift+P > Raspberry Pi Pico > New Project` ?

Octopus1633, 2024-09-09:
> Hi, do you have any output in the "Output > Extension Host" panel? and do you have a small example project on GitHub that silently fails when importing?

Thanks for your reply. I solved this problem later. It was because my CMakeLists.txt file contained Chinese comments (it was indeed comments). I changed them to English comments or deleted them and it was successfully imported!

will-v-pi, 2024-10-31:
> What is the recommended way of importing a project, and are there any rules about what sort of projects can be imported?

If a project has multiple executables, or the project name is defined as a variable, then you'll need to enable the CMake Tools integration when importing as detailed in the [README](https://github.com/raspberrypi/pico-vscode/tree/0.17.0?tab=readme-ov-file#cmake-tools-extension-integration) - this should work from the latest version or the extension published today.

Ideally it should be possible to import any pico project, so if you come across a project where importing doesn't work with the latest debug development version of the extension (the latest .vsix can be downloaded from the Actions tab) then let us know and we'll see if we can make it work. Always test with the debug development verison, and provide the logs from the developer console, as that will allow us to see what's going wrong.

> I have tried to import several projects, and often see the same failure: after executing the `Import Project` command, the OUTPUT window shows the following error:
> 
> ```
> [10/31/2024, 2:04:29 PM] "${workspaceFolder}/build/compile_commands.json" could not be parsed. 'includePath' from c_cpp_properties.json in folder 'rnnoise-examples-for-pico-2' will be used instead.
> ```
> 
> Indeed, if you look, there is no `compile_commands.json` file generated in the build directory.

This file is only created if `CMAKE_EXPORT_COMPILE_COMMANDS` is set in the CMake project. Most imported projects won't have this line, so will get this warning, but it gets set automatically by the CMake Tools extension, so if you're using that integration the warning will go away.

This warning just relates to Intellisense, so will not impact project compilation at all, and the includePath used instead will work perfectly fine in most cases so the warning can be ignored.

> This particular example uses the repository `https://github.com/ArmDeveloperEcosystem/rnnoise-examples-for-pico-2`, which was featured in the Raspberry Pi blog post [Real-time ML audio noise suppression on Raspberry Pi Pico 2](https://www.raspberrypi.com/news/real-time-ml-audio-noise-suppression-on-raspberry-pi-pico-2/), which I would like to use.
> 
> I assume that this particular example was built using the older SDK, but if that's the case what is the recommended way of importing? Setting the SDK to 1.5.1 or 1.5.0 on the import settings panel doesn't make any difference.

If that example is for the Pico 2 then it'll be using the 2.0.0 SDK - that is the only SDK that support RP2350. I've just tested importing that project with the new extension (0.17.0) and it works fine with the CMake Tools integration enabled, so I'd recommend giving the new version of the extension a try.

> What about the `pico_sdk_import.cmake` file? will this always be recreated based on my installed SDK? I have tried deleting this file before importing, and it seems to be recreated correctly, but doesn't change the behaviour

That file will be overwritten with the one for the chosen SDK when importing, but it doesn't actually matter - the only bit in that file that's ever changed is the fetching from Git, which this extension never uses, so an older version of that file should not cause any problems.

byzantic, 2024-10-31:
Thanks for the tips. I updated to the latest pico extension release, and enabled the CMake Tools integration during import.

I _did_ get it  building eventually, though I had to manually fetch some git submodules. I encountered various issues along the way, which I solved in the time-honoured manner of just pressing knobs until something different happened.

I think the majority of issues are due to the original project structure, rather than the extension. 

Thanks again!

closed 2025-01-15 by will-v-pi