raspberrypi/pico-vscode #64: Generating RISC-V project fails

Vinzenz82, 2024-08-31:
When I enable the RISC-V checkbox in "New Project" view the project generation fails. 

Seems something can not be loaded:
console.ts:137 [Extension Host] [raspberry-pi-pico] SDK 2.0.0 is already installed.
console.ts:137 [Extension Host] [raspberry-pi-pico] Error while downloading Toolchain: Not Found

Full Log:

```javascript
main.js:493 Updated selected toolchain with new default value RISCV_RPI
main.js:506 Uncaught TypeError: Cannot read properties of undefined (reading 'checked')
    at _onMessage (main.js:506:35)
_onMessage @ main.js:506
postMessage (async)
(anonymous) @ index.html?id=422b72cd-dbcd-4cec-8c55-469d2185079d&origin=c6102ada-87bd-43c6-9250-7866f55162ef&swVersion=4&extensionId=raspberry-pi.raspberry-pi-pico&platform=electron&vscode-resource-base-authority=vscode-resource.vscode-cdn.net&parentOrigin=vscode-file%3A%2F%2Fvscode-app:1197
HostMessaging.channel.port1.onmessage @ index.html?id=422b72cd-dbcd-4cec-8c55-469d2185079d&origin=c6102ada-87bd-43c6-9250-7866f55162ef&swVersion=4&extensionId=raspberry-pi.raspberry-pi-pico&platform=electron&vscode-resource-base-authority=vscode-resource.vscode-cdn.net&parentOrigin=vscode-file%3A%2F%2Fvscode-app:321
event.ts:42 [Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
onWillAddFirstListener @ event.ts:42
q @ event.ts:1137
(anonymous) @ event.ts:127
onWillAddFirstListener @ event.ts:202
q @ event.ts:1137
K @ listWidget.ts:694
D @ listWidget.ts:1547
se @ listWidget.ts:1527
Z @ listService.ts:288
o @ instantiationService.ts:162
createInstance @ instantiationService.ts:128
r @ notificationsList.ts:74
show @ notificationsList.ts:53
Q @ notificationsToasts.ts:213
(anonymous) @ notificationsToasts.ts:160
execute @ dom.ts:326
mt @ dom.ts:367
(anonymous) @ dom.ts:385
event.ts:42 [Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
onWillAddFirstListener @ event.ts:42
q @ event.ts:1137
(anonymous) @ event.ts:127
onWillAddFirstListener @ event.ts:202
q @ event.ts:1137
K @ listWidget.ts:694
D @ listWidget.ts:1547
se @ listWidget.ts:1527
Z @ listService.ts:288
o @ instantiationService.ts:162
createInstance @ instantiationService.ts:128
r @ notificationsList.ts:74
show @ notificationsList.ts:53
Q @ notificationsToasts.ts:213
(anonymous) @ notificationsToasts.ts:160
execute @ dom.ts:326
mt @ dom.ts:367
(anonymous) @ dom.ts:385
console.ts:137 [Extension Host] [raspberry-pi-pico] SDK 2.0.0 is already installed.
console.ts:137 [Extension Host] [raspberry-pi-pico] Error while downloading Toolchain: Not Found
log.ts:439   ERR [Extension Host] [[31mERROR[0m] [NewProjectPanel] Failed to download and install toolchain and SDK.
console.ts:137 [Extension Host] [ERROR] [NewProjectPanel] Failed to download and install toolchain and SDK.
y @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
S @ rpcProtocol.ts:458
Q @ rpcProtocol.ts:443
M @ rpcProtocol.ts:373
L @ rpcProtocol.ts:299
(anonymous) @ rpcProtocol.ts:161
B @ event.ts:1230
fire @ event.ts:1261
fire @ ipc.net.ts:652
K.onmessage @ localProcessExtensionHost.ts:378
notificationsAlerts.ts:42 Failed to download and install SDK and/or toolchain.
c @ notificationsAlerts.ts:42
(anonymous) @ notificationsAlerts.ts:28
B @ event.ts:1230
C @ event.ts:1241
fire @ event.ts:1265
addNotification @ notifications.ts:228
notify @ notificationService.ts:253
(anonymous) @ mainThreadMessageService.ts:86
f @ mainThreadMessageService.ts:51
$showMessage @ mainThreadMessageService.ts:45
S @ rpcProtocol.ts:458
Q @ rpcProtocol.ts:443
M @ rpcProtocol.ts:373
L @ rpcProtocol.ts:299
(anonymous) @ rpcProtocol.ts:161
B @ event.ts:1230
fire @ event.ts:1261
fire @ ipc.net.ts:652
K.onmessage @ localProcessExtensionHost.ts:378
event.ts:42 [Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
onWillAddFirstListener @ event.ts:42
q @ event.ts:1137
(anonymous) @ event.ts:127
onWillAddFirstListener @ event.ts:202
q @ event.ts:1137
K @ listWidget.ts:694
D @ listWidget.ts:1547
se @ listWidget.ts:1527
Z @ listService.ts:288
o @ instantiationService.ts:162
createInstance @ instantiationService.ts:128
r @ notificationsList.ts:74
show @ notificationsList.ts:53
Q @ notificationsToasts.ts:213
(anonymous) @ notificationsToasts.ts:160
execute @ dom.ts:326
mt @ dom.ts:367
(anonymous) @ dom.ts:385
```

will-v-pi, 2024-09-02:
Yes, this is a bug in the current release, which has been fixed in d931311f1671fc86ee0a040c16db412f131684f7 - the fix will be in the next release

will-v-pi, 2024-09-11:
This has been fixed in the 0.16.0 release

closed 2024-09-11 by will-v-pi