raspberrypi/pico-vscode #66: Severely broken compilation after performing CMake: Clean for a Pico 2 project.

nyh-workshop, 2024-09-01:
Hi Raspberry Pi team,

I encountered a severely broken compilation after doing "CMake: Clean" (Ctrl-shift-P and you can find that command) on a Pico 2 project with ARM cores.  

The compile errors range from being defaulted back to "RP2040", to a lot of other linker errors. Here is this primary example:

```
ninja: Entering directory `E:\pico-projects-2024\pico2dexed/build'
[0/1] Re-running CMake...PICO_SDK_PATH is C:/Users/yongh/.pico-sdk/sdk/2.0.0
Target board (PICO_BOARD) is 'pico'.
Using board configuration from C:/Users/yongh/.pico-sdk/sdk/2.0.0/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
```

Looks like the CMakeLists.txt has this one missing at the beginning of the file so I added it back again:
```
set(PICO_BOARD pico2 CACHE STRING "Board type")
```

However, after doing "Delete Cache and Reconfigure", it just doesn't compile anymore and now presenting me with more linker errors:

```
C:/Users/yongh/.pico-sdk/toolchain/13_2_Rel1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/pico2dexed.dir/src/sounddevice.cpp.obj: in function `CSoundDevice::Init(unsigned long)':
E:/pico-projects-2024/pico2dexed/src/sounddevice.cpp:26:(.text._ZN12CSoundDevice4InitEm+0x32): undefined reference to `__cxa_guard_acquire'
C:/Users/yongh/.pico-sdk/toolchain/13_2_Rel1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: E:/pico-projects-2024/pico2dexed/src/sounddevice.cpp:26:(.text._ZN12CSoundDevice4InitEm+0x40): undefined reference to `__cxa_guard_release'
C:/Users/yongh/.pico-sdk/toolchain/13_2_Rel1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: synth_dexed/libsynth_dexed.a(PluginFx.cpp.obj): in function `PluginFx::init(unsigned short)':
E:/pico-projects-2024/pico2dexed/synth_dexed/Synth_Dexed/src/PluginFx.cpp:74:(.text._ZN8PluginFx4initEt+0x4c): undefined reference to `sqrtf'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

As a result I have to delete the entire project and re-generate the Pico project and manually add the files I worked on earlier to compile it successfully again.

I have a snapshot of the failed build and the fresh build at my new project in Git: https://github.com/nyh-workshop/pico2dexed/commits/main/

The first two commits and by some comparisons, seems that the build had defaulted back to RP2040: 
* 80bf61d : original build that failed compilation after CMake: Clean and Delete Cache and Reconfigure
* 7b42635 : project has been deleted and re-generated again (compilation is working as usual).

nyh-workshop, 2024-09-01:
On top of that, do we have a "gitignore" for each Pico-SDK app generated? I'm not sure what to gitignore the compile intermediates at the build folder there.

paulober, 2024-09-01:
Hi @nyh-workshop, thanks for reporting the issue.
I have some questions that can help us track down the issue.
- First, does this also happen for you if you now generate a new project with the latest version of the extension and run the CMake clean command?
- Does following file exist on your system: `%USERHOME%/.pico-sdk/cmake/pico-vscode.cmake`?

For configuring of the build system after deleting/cleaning the build folder I generally recommend using our built-in `Raspberry Pi Pico: Configure CMake` command.

Concerning the question what to ignore on a generated pico-sdk app: you should exclude the full `build/` folder as in the cmake configure process it will be filled with platform specific ninja build configuration files therefore syncing this directory only causes problems on different system and wast space on your version control system.

nyh-workshop, 2024-09-02:
Hi @paulober , thanks for the support.

* This doesn't happen if I create a fresh new project, and then do a "CMake: Clean", and I recompile it. It works as usual and links to the final binary file.
*  Yes, this pico-vscode.cmake is inside the `C:\Users\<my username>\.pico-sdk\cmake\pico-vscode.cmake`.

I am excluding and deleting the build folder and try doing the `Raspberry Pi Pico: Configure CMake` again.

nyh-workshop, 2024-09-02:
Hmm, I'm not sure if this output is correct or the compilation is being completed when I did this now:
 
1. Delete build folder
2. Raspberry Pi Pico: Configure CMake (I do not touch the `CMake: Clean` anymore)
3.  Compile pico2

Output (you can ignore the warning - the code was still under testing):
```
ninja: Entering directory `E:\pico-projects-2024\pico2dexed/build'
[21/174] Building C object synth_dexed/CMakeFiles/synth_dexed.dir/arm_math.c.obj
E:/pico-projects-2024/pico2dexed/synth_dexed/arm_math.c: In function 'arm_biquad_cascade_df1_init_f32':
E:/pico-projects-2024/pico2dexed/synth_dexed/arm_math.c:14:14: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers] 
   14 |   S->pCoeffs = pCoeffs;
      |              ^
[174/174] Linking CXX executable pico2dexed.elf
 *  Terminal will be reused by tasks, press any key to close it.
```

The `CMake: Clean` broke the compilation when I press this again, but can be corrected by doing what is suggested by you too. :D

paulober, 2024-09-02:
Great that this works for you.
Yes, the output seems correct to me. This should give you a `pico2dexed.elf` and `pico2dexed.uf2` in your build folder.
The selection if you compile for pico2 or a different board also only happens during the cmake configuration step. Afterwards it is
"hardcoded" in the build configurations files generated into your build folder.

paulober, 2024-09-02:
So CMake clean should not have been the issue but the configuration step after this requires the right environment configuration. Therefor we recommend using our cmake configure command otherwise you would have to setup cmake tools extension with the same settings.

closed 2024-09-02 by paulober

nyh-workshop, 2024-09-03:
Thanks for the help @paulober ! Now I can have cleaner Pico2 repositories! :D