raspberrypi/pico-vscode #68: support debugging multiple executables within a project

jdbruner, 2024-09-02:
When a project has multiple executables, there is apparently no way to specify which one will be loaded or debugged. As an example, the `picow_blink` example has two executables: `picow_blink` and `pico_blink_slow_clock`. However, the "Debug Project" operation appears to assume that the `project` in `CMakeLists.txt` is also the name of the executable, so it will always run/debug `pico_blink_slow_clock`. There appears to be no way to debug any other executable within the project (in this case `picow_blink`).

The CMake VSCode extension does support running/debugging of all executables that are defined within the project. This can be used by settings the `raspberry-pi-pico.useCmakeTools` workspace setting to `true`. (The setting for the extension does not seem to be propagated into new workspaces, so this must be done every time, along with a couple of additional configuration steps so that it will build properly.)

Please consider adding to the pico-vscode extension an option to select the executable to be run/debugged in a way similar to what the CMake extension supports.

will-v-pi, 2024-09-02:
If you set that `useCmakeTools` variable to `true` and then try debugging using the methods in `launch.json` added by the extension, it should allow for the debugging of multiple executables.

By default the extension does not use this as the CMake Tools extension throws a lot of different messages which can be confusing for new users, but that variable switches it to using that extension for compiling and selecting the target for debugging, so it should work for your use case

jdbruner, 2024-09-02:
OK, I understand why this is `false` by default for the extension, but I was a little surprised that new workspaces don't inherit the extension setting. Turning it on for the extension has no effect - it is still necessary to turn it on in each workspace. It might be good to mention this in the setting description.

will-v-pi, 2024-09-25:
We've added more documentation on the cmake tools integration to the readme now, and an option to enable it when creating/importing a project.

closed 2024-09-25 by will-v-pi