raspberrypi/pico-vscode #70: Add experimental toolchain

peardox, 2024-09-03:
I've built a functional toolchain for Win64 / Pico 2 / RiscV-Hazard as per https://github.com/Wren6991/Hazard3/issues/16

The toolchain tests out fine and I can build a uf2 and stick it on a Pico 2 but it'd be 'smoother' if we could add our own toolchains to the extension

I had a look around the extension but adding something like....

> [RISCV_HAZARD3_SEP_24]
> win32_x64 = https://peardox.com/pico/gcc14-no-zcmp-no-multilib.zip
>

To supportedToolchains.ini and setting my toolchain to RISCV_HAZARD3_SEP_24 failed to download my build

I also tried changing "riscvToolchain": "RISCV_COREV_MAY_24" to suit in versionBundles.json

Manually installing the experimental toolchain then building a RiscV project and manually altering and building via msys/mingw64 did, however, work OK and produced functional results.

will-v-pi, 2024-09-03:
I think this is not a priority, as if you're able to compile a custom toolchain on Windows then you're probably able to manually set up your Pico SDK environment. But we might look into adding the option to select a custom toolchain path in the future.

Although I'd be happy to take any suggestions on how to build a Risc-V toolchain on Windows in the raspberrypi/pico-sdk-tools repo - currently I'm taking a prebuilt toolchain from elsewhere, and copying in a multilib built on MacOS which has been compiled with all the relevant extensions. If you have a way to build natively on Windows then I could try switching to that - I had issues with `texinfo` when I tried it myself

will-v-pi, 2025-01-15:
Closing this as not planned, as you can just manually set your own `PICO_TOOLCHAIN_PATH` and `PICO_PLATFORM` in CMakeLists.txt, to override the toolchain used by this extension and use a custom toolchain instead. You would need to do this after the `DO NOT EDIT` block and before the `include(pico_sdk_import.cmake)`.

closed 2025-01-15 by will-v-pi