raspberrypi/pico-vscode #76: Error: Failed to find python3 executable - New install of SDK with Codium aka VSCodium

Andy2No, 2024-09-08:
On reaching step 6 of Chapter 4 of the getting started guide ("Click Create to create the project."), I get two error boxes:

1: Failed to find python3 executable

2: Generating project based on the blink example in c:\Users\andy\Codium, this may take a while...: Failed

I have python3.8.6 installed, and it's on the command path.  It's in C:\Program Files\Python38

Where is it trying to look for python3 and what is it expecting to see there so that it knows it has found it?  Is there a setting I can change to tell it where to look?

I'm trying to set up the SDK with Codium, instead of VSCode, following the official Getting Started guide for all other details.  I'm using the latest release of Codum for Windows x64, which is Codium_Setup_1.92.2.24235_win64.exe, with a manually installed Raspberry Pi Pico Visual Studio Code extension 0.15.1 from https://open-vsx.org/extension/raspberry-pi/raspberry-pi-pico.

will-v-pi, 2024-09-08:
The version on OpenVSX should be 0.15.2 - can you double check if your version is 0.15.2 or 0.15.1?

If you click “Show advanced options” at the bottom of the page (next to the Create button) it will reveal a Python installation selector - you can specify a full path to Python there. Although by default the extension should download its own copy of Python 3.12, so I’m not sure why that’s failing to happen in your case? Could you open the developer console (Help -> Toggle Developer Tools) and post the output that you get there?

In windows the extension looks for a `python` executable in the path, not a `python3` executable, so it’s also possible that’s the problem

Andy2No, 2024-09-08:
Thanks.  I tried 0.15.2 first but it just showed an error, when clicking on the extension on the left, in Explorer view:

"There is no data provider registered that can provide view data"

That was in place of the menu which includes "New project from examples".  Version does 0.15.1 provide that menu so that seemed like the best choice.

I'm trying to set this up on a Windows 7 machine, with the goal of having the same environment set up on all my PCs, if possible, which is why I've chosen Codium over VSCode.  Win 7 is still the most stable and user friendly version of Windows I currently have, mostly due to there being no more updates for it.  It works fine with Arduino 1.8.19 and Earle Philhower's arduino-pico core, but so far, I haven't managed to set up the pico sdk on it.

AFAIK, python 3.12 isn't supported, and 3.8.6 might be the final version that is - not entirely sure about that part.

Using Advanced options, I'm able to select "Path to executable" then click "Choose files" to specify C:\Program Files\Python38\python.exe .  The first time I did that then pressed Create, it looked busy for a while, then gave a failure notification.  The second time, it did nothing other than go to the home/main page, with a red bar along the bottom.  Clicking the notifications icon now says "NO NEW NOTIFICATIONS".

I found some settings for the extension earlier, via File->Preferences->Settings, with one that says "Path to the Python executable [no spaces]".  That makes me wonder whether having a space in the path is a problem when set via advanced settings.

Here is the current contents of the developers console.  I don't know how to tell which part is new or was added when:

```
workbench.desktop.main.js:660  WARN Settings pattern "application.*" doesn't match any settings
workbench.desktop.main.js:660  INFO Started local extension host with pid 7020.
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.armToolchainPath.linux as cortex-debug.armToolchainPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.armToolchainPath.osx as cortex-debug.armToolchainPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.armToolchainPath.windows as cortex-debug.armToolchainPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.JLinkGDBServerPath.linux as cortex-debug.JLinkGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.JLinkGDBServerPath.osx as cortex-debug.JLinkGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.JLinkGDBServerPath.windows as cortex-debug.JLinkGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.openocdPath.linux as cortex-debug.openocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.openocdPath.osx as cortex-debug.openocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.openocdPath.windows as cortex-debug.openocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.pyocdPath.linux as cortex-debug.pyocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.pyocdPath.osx as cortex-debug.pyocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.pyocdPath.windows as cortex-debug.pyocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.PEGDBServerPath.linux as cortex-debug.PEGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.PEGDBServerPath.osx as cortex-debug.PEGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.PEGDBServerPath.windows as cortex-debug.PEGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stutilPath.linux as cortex-debug.stutilPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stutilPath.osx as cortex-debug.stutilPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stutilPath.windows as cortex-debug.stutilPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stlinkPath.linux as cortex-debug.stlinkPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stlinkPath.osx as cortex-debug.stlinkPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stlinkPath.windows as cortex-debug.stlinkPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stm32cubeprogrammer.linux as cortex-debug.stm32cubeprogrammer is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stm32cubeprogrammer.osx as cortex-debug.stm32cubeprogrammer is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stm32cubeprogrammer.windows as cortex-debug.stm32cubeprogrammer is null
workbench.desktop.main.js:660  WARN Settings pattern "application.*" doesn't match any settings
workbench.desktop.main.js:97 [Extension Host]% Object
workbench.desktop.main.js:97 [Extension Host]% Object
workbench.desktop.main.js:97 [Extension Host] [raspberry-pi-pico] Extension activated.
workbench.desktop.main.js:97 [Extension Host] [INFO] [UI] Initializing UI
workbench.desktop.main.js:97 [Extension Host] [raspberry-pi-pico] No workspace folder or pico project found.
workbench.desktop.main.js:97 [Extension Host]% Object
workbench.desktop.main.js:97 [Extension Host]% Object
workbench.desktop.main.js:660  WARN Settings pattern "application.*" doesn't match any settings
workbench.desktop.main.js:660  WARN SettingsEditor2: Settings not included in settingsLayout.ts: issueReporter.experimental.auxWindow
[... cut ...]
workbench.desktop.main.js:97 [Extension Host] [raspberry-pi-pico] Successfully downloaded examples list from the internet.
workbench.desktop.main.js:97 [Extension Host] [INFO] [NewProjectPanel] Loaded 84 examples.
tailwindcss-3_3_5.js:62 cdn.tailwindcss.com should not be used in production. To use Tailwind CSS in production, install it as a PostCSS plugin or use the Tailwind CLI: https://tailwindcss.com/docs/installation
(anonymous) @ tailwindcss-3_3_5.js:62
(anonymous) @ tailwindcss-3_3_5.js:62
main.js:427 set theme dark
```

will-v-pi, 2024-09-08:
I’m afraid this extension is only currently supported on Windows 10 & 11, as those are the versions that VS Code supports. Windows 7 is not supported as it reached end of life in 2020 and no longer receives security updates, and VS Code stopped supporting it in 2022.

Andy2No, 2024-09-08:
I'm aware that VSCode support for Windows 7 has ended.  Codium is a build of VSCode without telemetry and still supports Windows 7 - it's specifically discussed in the repo for it.  It appears to work fine.

I noticed there are a couple of issues open for v0.15.2 of the extension, e.g. #73 and #54 which seem maybe related but didn't seem to be an exact match for what I experienced.  Since they're all beta releases, there didn't seem to be any harm in trying v0.15.1, which at least gives me the user interface to try it, instead of just an error message.

I get the impression that this is largely a problem of configuring things in the extension. I don't have any insight into what the developer console log says, and the error messages I've been given by it aren't really helpful.  It started with saying that Git wasn't installed or on the path, neither of which was true, assuming it meant the PATH environment variable.  The solution I found for that was to uninstall Git then install a newer version - which took me to the not entirely helpful error message about the missing Python.  Without understanding the error messages or developer logs, I can't know for certain that it can't work; I just don't really get any clues from it about what the problems actually are.

will-v-pi, 2024-09-08:
The UI issues with 0.15.2 that you’re seeing are actually the extension failing to activate, which is a more fundamental issue than the configuration issues - I think the best bet for debugging would be to use 0.15.2 and post the output from the developer console and also the extension host when you try to open the extension, but my guess would be that one of the extension’s dependencies doesn’t support Windows 7, and therefore the extension fails to activate

I’ll add a note to the readme that only Windows 10 & 11 are currently supported. We’d be happy to take a PR adding Windows 7 support (provided it isn’t detrimental to other functionality), but I don’t think we’ll be able to support it ourselves, as we don’t have internet connected Windows 7 systems to test on due to them being insecure

Andy2No, 2024-09-08:
Thanks.  Here's the Developers tools output, starting after uninstalling 0.15.1.  After turning the Developers tools on, I then tried to install 0.15.2.  I don't know how to get the extension host part because I'm not sure what that means.  Codium reports the installation as successful but then still shows:
"There is no data provider registered that can provide view data"

```
workbench.desktop.main.js:660  INFO Started local extension host with pid 3228.
workbench.desktop.main.js:660  WARN [twxs.cmake]: Cannot register 'cmake.cmakePath'. This property is already registered.
workbench.desktop.main.js:97 [Extension Host]% Object
workbench.desktop.main.js:660  INFO [perf] Render performance baseline is 54ms
DevTools failed to load source map: Could not load content for https://main.vscode-cdn.net/sourcemaps/adc83b19e793491b1c6ea0fd8b46cd9f32e592fc/core/vs\workbench\workbench.desktop.main.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE
DevTools failed to load source map: Could not load content for https://main.vscode-cdn.net/sourcemaps/adc83b19e793491b1c6ea0fd8b46cd9f32e592fc/core/vs\code\electron-sandbox\workbench\workbench.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE
DevTools failed to load source map: Could not load content for https://main.vscode-cdn.net/sourcemaps/adc83b19e793491b1c6ea0fd8b46cd9f32e592fc/core/vs\base\parts\sandbox\electron-sandbox\preload.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.armToolchainPath.linux as cortex-debug.armToolchainPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.armToolchainPath.osx as cortex-debug.armToolchainPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.armToolchainPath.windows as cortex-debug.armToolchainPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.JLinkGDBServerPath.linux as cortex-debug.JLinkGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.JLinkGDBServerPath.osx as cortex-debug.JLinkGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.JLinkGDBServerPath.windows as cortex-debug.JLinkGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.openocdPath.linux as cortex-debug.openocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.openocdPath.osx as cortex-debug.openocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.openocdPath.windows as cortex-debug.openocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.pyocdPath.linux as cortex-debug.pyocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.pyocdPath.osx as cortex-debug.pyocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.pyocdPath.windows as cortex-debug.pyocdPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.PEGDBServerPath.linux as cortex-debug.PEGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.PEGDBServerPath.osx as cortex-debug.PEGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.PEGDBServerPath.windows as cortex-debug.PEGDBServerPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stutilPath.linux as cortex-debug.stutilPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stutilPath.osx as cortex-debug.stutilPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stutilPath.windows as cortex-debug.stutilPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stlinkPath.linux as cortex-debug.stlinkPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stlinkPath.osx as cortex-debug.stlinkPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stlinkPath.windows as cortex-debug.stlinkPath is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stm32cubeprogrammer.linux as cortex-debug.stm32cubeprogrammer is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stm32cubeprogrammer.osx as cortex-debug.stm32cubeprogrammer is null
workbench.desktop.main.js:660   ERR Ignoring cortex-debug.stm32cubeprogrammer.windows as cortex-debug.stm32cubeprogrammer is null
workbench.desktop.main.js:97 [Extension Host]% {}
workbench.desktop.main.js:2096 Activating extension 'raspberry-pi.raspberry-pi-pico' failed: ReadableStream is not defined.
$onExtensionActivationError @ workbench.desktop.main.js:2096
_doInvokeHandler @ workbench.desktop.main.js:1909
_invokeHandler @ workbench.desktop.main.js:1909
_receiveRequest @ workbench.desktop.main.js:1909
_receiveOneMessage @ workbench.desktop.main.js:1909
(anonymous) @ workbench.desktop.main.js:1909
_deliver @ workbench.desktop.main.js:93
fire @ workbench.desktop.main.js:93
fire @ workbench.desktop.main.js:123
G.onmessage @ workbench.desktop.main.js:1925
```

will-v-pi, 2024-09-09:
> workbench.desktop.main.js:2096 Activating extension 'raspberry-pi.raspberry-pi-pico' failed: ReadableStream is not defined.

This looks like you have an incompatible NodeJS version as the [`ReadableStream`](https://nodejs.org/api/webstreams.html#class-readablestream) class was added in v16.5.0, but the latest NodeJS version that (officially) supports Windows 7 is v13.6.0. You can check what NodeJS version you have by clicking Help -> About - I'm not sure what version Electron/Codium uses for Windows 7 builds, but it's probably too old for this extension

Andy2No, 2024-09-09:
It seems I have two versions of NodeJS installed.  If I start it from the Programs menu, it says "Welcome to Node.js v18.20.2.", but if  I look at Help->About in Codium, it shows 16.17.1.

I made two separate attempts at installing Codium because I initially missed seeing the latest releases.  I think the higher number was from the first, unsuccessful attempt.

The older installer launched a script to install other things, which got a bit confusing.  The latest one was much simpler, and I guess that has installed the NodeJS 16.7.1 along with it.  I'm under the impression that the NodeJS used is also built specifically to run on Windows 7, but I don't fully understand the installation process for it all.

Here's a screenshot of the About dialog box.

[image]

Maybe I should uninstall Codium and the other NodeJS and start again?

closed 2025-03-12 by will-v-pi