raspberrypi/pico-vscode #80: Accessing all examples?

chupocro, 2024-09-10:
Is there a way of accessing all the examples from the [repository](https://github.com/raspberrypi/pico-examples) by using `New Project From Example`? I'd like to test some `PIO` examples but there isn't any in the drop-down menu.

I am not sure if my `New Project From Example` window works as it should because if the `Board type` is `Pico W` then there are `Basic Settings`, `Pico wireless options` and `Debugger` tabs at the top but clicking on `Pico wireless options` or `Debugger` has no effect.

Blink example with Pico board works well.

paulober, 2024-09-11:
So the Pico W issue is already fixed and will be included in the next release. So your examples view is working correctly only these links shouldn't pop up.

No unfortunately not. The extension includes a custom reduced set of examples for better integration with the project generator.

So the current workflow would be to just clone or download the examples repo once and then use the import project dialog and open the example you want to use.

This also requires adding some additional statements to the CMakeLists.txt in there so it can compile correctly. I may add an option to the import process to optimize for importing pico examples.

will-v-pi, 2024-09-11:
The PIO examples have been added in the 0.16.0 release, so closing this issue

closed 2024-09-11 by will-v-pi