raspberrypi/pico-vscode #82: New Version 0.16.0 Breaks Project Creation, Import

gov-sprymon, 2024-09-12:
As of yesterday(9/11/2024), the New Project, Import Project, New Project From Example are broken. 

The only error I see "Could not create new project: <project name>"

Linux  6.1.0-25-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.106-3 (2024-08-26) x86_64 GNU/Linux

cdmuhlb, 2024-09-13:
I also encountered this problem when following the '[getting started](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)' guide to create a "blink" project with version 0.16.0.  Going to "Developer: Show Logs... | Window", we see the error:
```
2024-09-12 22:33:43.952 [error] [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - NewProjectPanel] Generator Process exited with code: 1
2024-09-12 22:33:43.955 [error] [Extension Host] [[31mERROR[0m] [raspberry-pi-pico - NewProjectPanel] Generator Process error: Command failed: "python3" "/home/user/.vscode/extensions/raspberry-pi.raspberry-pi-pico-0.16.0/scripts/pico_project.py" -board pico -exam -d 0 --project vscode --projectRoot "/home/user/Projects" --sdkVersion 2.0.0 --toolchainVersion 13_2_Rel1 --picotoolVersion 2.0.0 --openOCDVersion 0.12.0+dev --ninjaPath "/home/user/.pico-sdk/ninja/v1.12.1/ninja" --cmakePath "/home/user/.pico-sdk/cmake/v3.28.6/bin/cmake"  "blink"
  File "/home/user/.vscode/extensions/raspberry-pi.raspberry-pi-pico-0.16.0/scripts/pico_project.py", line 873
    '''
       ^
SyntaxError: f-string expression part cannot include a backslash
```
My Python version is 3.10.12.

**This appears to be a regression introduced by PR #72** (two days before the release of 0.16.0).  I have confirmed that I can create the project successfully after downgrading to version 0.15.2.

It looks like the `debugServerArgs` property of the "Pico Debug (C++ Debugger)" JSON object needs to be produced using a different string interpolation method, to avoid backslashes in format string brace substitution expressions.

will-v-pi, 2024-09-13:
Thanks for raising this - should be fixed with that commit, and we'll do a bug-fix release shortly

closed 2024-09-13 by will-v-pi

will-v-pi, 2024-09-13:
Release 0.16.1 is out now, so that should fix this

cdmuhlb, 2024-09-13:
I confirm that I can create new projects after upgrading to 0.16.1.  Thank you for the quick fix!

commit 72a02fce (2024-09-13): Fix #82 - f-string expression part cannot include a backslash