raspberrypi/pico-vscode #83: FYI: missing libs for openocd on macOS 14.6.1: libftdi and libcapstone

mchapman87501, 2024-09-15:
On macOS, `openocd` has runtime dependencies on `libftdi` and `libcapstone`.  I needed to install them manually using `brew install`.

Before manual installation:
```shell
$ ls -lart ~/.pico-sdk/openocd/0.12.0+dev 
total 9320
drwxr-xr-x   3 my_username  staff       96 Sep  2 18:28 ..
-rw-r--r--   1 my_username  staff    76592 Sep  2 18:28 libhidapi.dylib
-r--r--r--   1 my_username  staff   179152 Sep  2 18:28 libusb-1.0.dylib
-rwxr-xr-x   1 my_username  staff  4511000 Sep  2 18:28 openocd
drwxr-xr-x  15 my_username  staff      480 Sep  2 18:28 scripts
lrwxr-xr-x   1 my_username  staff       59 Sep  2 18:28 openocd.exe -> /Users/my_username/.pico-sdk/openocd/0.12.0+dev/openocd
$ otool -L /Users/my_username/.pico-sdk/openocd/0.12.0+dev/openocd
/Users/my_username/.pico-sdk/openocd/0.12.0+dev/openocd:
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
	@loader_path/libusb-1.0.dylib (compatibility version 5.0.0, current version 5.0.0)
	@loader_path/libhidapi.dylib (compatibility version 0.0.0, current version 0.14.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1345.100.2)
	/opt/homebrew/opt/capstone/lib/libcapstone.5.dylib (compatibility version 5.0.0, current version 5.0.1)
$ ls -l /opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib
ls: /opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib: No such file or directory
$ ls -l /opt/homebrew/opt/capstone/lib/libcapstone.5.dylib
ls: /opt/homebrew/opt/capstone/lib/libcapstone.5.dylib: No such file or directory
```

will-v-pi, 2024-09-15:
Could you try deleting your `~/.pico-sdk/openocd/` folder and reloading your window, which will download the latest OpenOCD which doesn’t have these runtime dependencies? Make sure you have the latest version of the extension first (0.16.1)

mchapman87501, 2024-09-15:
That worked.  Thank you!

Details:
* Deleted `~/.pico-sdk/openocd/`
* Ran `brew uninstall libftdi capstone` for good measure
* Reloaded the window
 
`otool -L`  on the new `openocd` showed the dependencies were gone. `openocd` ran without issue.

will-v-pi, 2024-09-16:
Glad it’s working - closing this as done

closed 2024-09-16 by will-v-pi