raspberrypi/pico-vscode #84: Since 0.16.1 libgpiod.so.2 is missing

schnitzeltony, 2024-09-16:
Hi,

started playing around with this plugin on Fedora. To get team mates up fast we created some automatic setup [1] which worked perfectly on several machines. 

Now that VSCode updated plugin automatically, debugging stopped working. Diving into lead to:

`
/home/superandy/.pico-sdk/openocd/0.12.0+dev/openocd.exe: error while loading shared libraries: libgpiod.so.2
`

Tried to hack around by installing openocd from Fedora and do adjustments in launch.json to start Fedora's openocd.

That works for pico but does not work for pico2 - seems Fedora's openocd is not yet aware of pico2.

Ahh and deletion of `~/.pico-sdk/openocd/` as suggested in [2] did not help.

Suggestions welcome :)

[1] https://github.com/ZeraGmbH/fedora-setup/blob/master/setup-raspi-pico
[2] https://github.com/raspberrypi/pico-vscode/issues/83

will-v-pi, 2024-09-16:
Actually relates to #40 - you’ll need to install libgpiod with your system’s package manager, and then it should work.

Feel free to open a PR to add a note to the README, like we added for #40

schnitzeltony, 2024-09-16:
Forgot to mention: Have done that but got a libgpiod.so.3

schnitzeltony, 2024-09-17:
Tried pragmatic approach: 
Let my fedora-setup scripts install dependencies and build raspi-fork of openocd. Build works but debugging now complains with different error.
What configure options do you use to build openocd? Is this information public available?

schnitzeltony, 2024-09-17:
That turned into a blocker: your openocd fork seems to depend on libgpiod-1.x.x. Fedora ships libgpiod-2.x.x. So now I see compile errors for changed API

will-v-pi, 2024-09-17:
The build system is public in the raspberrypi/pico-sdk-tools repository. We recently added libgpiod to all Linux builds, but given this issue I think the solution is to only add libgpiod to arm64 Linux builds, as it's only used by the extension when running on Raspberry Pis. I'll push a fix and update the extension

schnitzeltony, 2024-09-17:
Ahh OK - that is good news - so I can stop hacking further

Thanks so far

closed 2024-09-17 by will-v-pi

schnitzeltony, 2024-09-17:
Thanks for super quick fix!

commit 09304902 (2024-09-17): Bump tools version to v2.0.0-4

Fixes #84