raspberrypi/pico-vscode #86: Version 0.16.2: Failed to download and install SDK and/or toolchain

LittleYe233, 2024-09-18:
I am using the latest VS Code Insiders and Arch Linux, and trying to create an RISC-V example project for Pico 2, but the extension complained that "Failed to download and install SDK and/or toolchain". The selected RISC-V toolchain version is RISCV.RPI.2.0.0.2. Downloading ARM toolchain is always successful. I have found some similar issues and tried to downgrade to something like v0.15.1, but it still failed. I tried to directly download the toolchain file defined in `data` folder of this repo, and it seemed accessible.

Here is the full log:
```
[Violation] 'setTimeout' handler took 50ms
console.ts:137 [Extension Host] [raspberry-pi-pico] Examples git ref is 7fe60d6b4027771e45d97f207532c41b1d8c5418

 (at console.<anonymous> (file:///opt/visual-studio-code-insiders/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:156:8542))
console.ts:137 [Extension Host] [raspberry-pi-pico] Sparse-checkout selected example: blink (at console.<anonymous> (file:///opt/visual-studio-code-insiders/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:156:8542))
console.ts:137 [Extension Host] [raspberry-pi-pico] Copying example from /home/littleye233/.pico-sdk/examples/blink to /run/media/littleye233/Data/code/Pico2 (at console.<anonymous> (file:///opt/visual-studio-code-insiders/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:156:8542))
console.ts:137 [Extension Host] [raspberry-pi-pico] Done copying example. (at console.<anonymous> (file:///opt/visual-studio-code-insiders/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:156:8542))
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - download] SDK 2.0.0 is already installed. (at console.<anonymous> (file:///opt/visual-studio-code-insiders/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:156:8542))
event.ts:42 [Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
onWillAddFirstListener @ event.ts:42
q @ event.ts:1149
(anonymous) @ event.ts:136
onWillAddFirstListener @ event.ts:211
q @ event.ts:1149
cEe @ listWidget.ts:690
D @ listWidget.ts:1543
ul @ listWidget.ts:1523
xg @ listService.ts:280
o @ instantiationService.ts:162
createInstance @ instantiationService.ts:128
r @ notificationsList.ts:74
show @ notificationsList.ts:53
Q @ notificationsToasts.ts:213
(anonymous) @ notificationsToasts.ts:160
execute @ dom.ts:326
n @ dom.ts:367
(anonymous) @ dom.ts:385
event.ts:42 [Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
onWillAddFirstListener @ event.ts:42
q @ event.ts:1149
(anonymous) @ event.ts:136
onWillAddFirstListener @ event.ts:211
q @ event.ts:1149
cEe @ listWidget.ts:690
D @ listWidget.ts:1543
ul @ listWidget.ts:1523
xg @ listService.ts:280
o @ instantiationService.ts:162
createInstance @ instantiationService.ts:128
r @ notificationsList.ts:74
show @ notificationsList.ts:53
Q @ notificationsToasts.ts:213
(anonymous) @ notificationsToasts.ts:160
execute @ dom.ts:326
n @ dom.ts:367
(anonymous) @ dom.ts:385
console.ts:137 [Extension Host] [INFO] [raspberry-pi-pico - download] Redirecting to https://objects.githubusercontent.com/github-production-release-asset-2e65be/839523001/5fb2b639-702c-4875-af9c-2d469afe6b74?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20240918%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240918T075916Z&X-Amz-Expires=300&X-Amz-Signature=1a3d65348da7f8b6ec8bf073bb1cf2ea6ad56f4cf9e26314d798dcf593f7839b&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=839523001&response-content-disposition=attachment%3B%20filename%3Driscv-toolchain-14-x86_64-lin.tar.gz&response-content-type=application%2Foctet-stream (at console.<anonymous> (file:///opt/visual-studio-code-insiders/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:156:8542))
log.ts:439   ERR [Extension Host] [ERROR] [raspberry-pi-pico - download] Error during download: other side closed
console.ts:137 [Extension Host] [ERROR] [raspberry-pi-pico - download] Error during download: other side closed (at console.<anonymous> (file:///opt/visual-studio-code-insiders/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:156:8542))
o7i @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
S @ rpcProtocol.ts:458
Q @ rpcProtocol.ts:443
M @ rpcProtocol.ts:373
L @ rpcProtocol.ts:299
(anonymous) @ rpcProtocol.ts:161
B @ event.ts:1242
fire @ event.ts:1273
fire @ ipc.net.ts:652
l.onmessage @ localProcessExtensionHost.ts:378
log.ts:439   ERR [Extension Host] [ERROR] [raspberry-pi-pico - download] Downloading or extracting Toolchain failed: other side closed
console.ts:137 [Extension Host] [ERROR] [raspberry-pi-pico - download] Downloading or extracting Toolchain failed: other side closed (at console.<anonymous> (file:///opt/visual-studio-code-insiders/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:156:8542))
o7i @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
S @ rpcProtocol.ts:458
Q @ rpcProtocol.ts:443
M @ rpcProtocol.ts:373
L @ rpcProtocol.ts:299
(anonymous) @ rpcProtocol.ts:161
B @ event.ts:1242
fire @ event.ts:1273
fire @ ipc.net.ts:652
l.onmessage @ localProcessExtensionHost.ts:378
log.ts:439   ERR [Extension Host] [ERROR] [raspberry-pi-pico - NewProjectPanel] Failed to download and install toolchain and SDK.
console.ts:137 [Extension Host] [ERROR] [raspberry-pi-pico - NewProjectPanel] Failed to download and install toolchain and SDK. (at console.<anonymous> (file:///opt/visual-studio-code-insiders/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:156:8542))
o7i @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
S @ rpcProtocol.ts:458
Q @ rpcProtocol.ts:443
M @ rpcProtocol.ts:373
L @ rpcProtocol.ts:299
(anonymous) @ rpcProtocol.ts:161
B @ event.ts:1242
fire @ event.ts:1273
fire @ ipc.net.ts:652
l.onmessage @ localProcessExtensionHost.ts:378
notificationsAlerts.ts:42 Failed to download and install SDK and/or toolchain.
c @ notificationsAlerts.ts:42
(anonymous) @ notificationsAlerts.ts:28
B @ event.ts:1242
C @ event.ts:1253
fire @ event.ts:1277
addNotification @ notifications.ts:228
notify @ notificationService.ts:253
(anonymous) @ mainThreadMessageService.ts:86
f @ mainThreadMessageService.ts:51
$showMessage @ mainThreadMessageService.ts:45
S @ rpcProtocol.ts:458
Q @ rpcProtocol.ts:443
M @ rpcProtocol.ts:373
L @ rpcProtocol.ts:299
(anonymous) @ rpcProtocol.ts:161
B @ event.ts:1242
fire @ event.ts:1273
fire @ ipc.net.ts:652
l.onmessage @ localProcessExtensionHost.ts:378
event.ts:42 [Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
onWillAddFirstListener @ event.ts:42
q @ event.ts:1149
(anonymous) @ event.ts:136
onWillAddFirstListener @ event.ts:211
q @ event.ts:1149
cEe @ listWidget.ts:690
D @ listWidget.ts:1543
ul @ listWidget.ts:1523
xg @ listService.ts:280
o @ instantiationService.ts:162
createInstance @ instantiationService.ts:128
r @ notificationsList.ts:74
show @ notificationsList.ts:53
Q @ notificationsToasts.ts:213
(anonymous) @ notificationsToasts.ts:160
execute @ dom.ts:326
n @ dom.ts:367
(anonymous) @ dom.ts:385
```

BTW, can I directly download the toolchain file and extract it to `~/.pico-sdk/toolchain/<RISCV-TOOLCHAIN-VERSION>`?

paulober, 2024-09-18:
Thanks for reporting the issue. Are you on arm64 or x86 (from the logs I assume x86)?

Yes, it should work if you download and extract it into `.pico-sdk\toolchain\RISCV_RPI_2_0_0_2` so the bin folder and everything is in this folder.

Can you try out the debug version of this release and see if you get any more logging output when reproducing the error?
https://github.com/raspberrypi/pico-vscode/releases/tag/0.16.2

paulober, 2024-09-18:
Can you try this debug build, it also includes a small change that might solve the issue for you: [raspberry-pi-pico-0.16.2.zip](https://github.com/user-attachments/files/17041581/raspberry-pi-pico-0.16.2.zip)
Have you setup a GitHub PAT for the extension?

LittleYe233, 2024-09-18:
@paulober Thank you for your fast reply! I have tested the debug version and it works as expected. (BTW I recommend to add a progress indicator in the console if possible, for the download operation may take long.)

I haven't set up a GitHub PAT yet because I thought it might not be very necessary. I don't know whether it affects the download operation or not.

paulober, 2024-09-18:
> I haven't set up a GitHub PAT yet because I thought it might not be very necessary. I don't know whether it affects the download operation or not.

That normally not required but it was just interesting for tracing down the issue.

> I have tested the debug version and it works as expected. (BTW I recommend to add a progress indicator in the console if possible, for the download operation may take long.)

Great. What do you mean by in the console? You currently should get a progress bar in the notifications area when downloading SDK and Toolchain.

closed 2024-09-18 by paulober

LittleYe233, 2024-09-18:
[image]
[image]

@paulober I have checked the console of developer tools and the pop-ups at the right bottom corner of VS Code (screenshots attached). I think there is nothing to indicate the detailed download progress like 10%, 20%, etc. If there is a progress bar elsewhere, please forgive my fault.

paulober, 2024-09-18:
I meant the one shown on your first screenshot.

commit fb3793c6 (2024-09-18): Fix #86, Fail to download on x86 Linux