raspberrypi/pico-vscode #87: Creating 'New Project From Example' fails with message 'Failed to setup example'

pointy56, 2024-09-18:
Having installed the 'pico' extension (v0.16.2) into VSCode 1.93.1 the icons for both the Raspberry Pi Pico project and the MicroPico Device Controller can be seen in the left-hand toolbar.
I select 'New Project From Example' from the Pi Pico Quick Access menu, then 'blink' and Pico, leaving the other settings to default.
When I click 'Create'  the extension issues the message 'Failed to setup example' - no other details are provided, but I am happy to provide any other information that you may require to investigate further.
I have tried on another system running the same version of VSCode and the new project is created successfully.

will-v-pi, 2024-09-18:
Do you have a version of git already installed, and if so what version is it? By default the extension downloads its own version of Git on Windows, but if you have a pre-existing version it will use that, however it requires a minimum git version of 2.28

pointy56, 2024-09-19:
I've just checked - the system that I'm having the problem with does indeed have git pre-installed, a very old version (2.10.1) - I guess it's not causing problems for anything else I use otherwise I would have updated it.
I'm happy to leave this version in-place for the time being if you would like to test a fix that handles this problem better.

will-v-pi, 2024-09-24:
This should be fixed with #93 - could you give that PR a go if you still have the old git on your system? No worries if not, I can test it myself too

You can download the .vsix file for that PR from https://github.com/raspberrypi/pico-vscode/actions/runs/11014326181?pr=93

PR #93 (2024-09-24): Add git version check before using system git
Check the git version is greater than 2.28.0 before using it, and throw an error on MacOS and Linux if the version is too low. On Windows it will download an up-to-date version instead and use that.

Fixes #87
closed 2024-09-24 by pointy56