raspberrypi/pico-vscode #88: Creating new "Project from Example" fails with "Failed to find python3 executable."

mtexplorative, 2024-09-19:
My set-up is a Mac, Apple-Silicon, Visual Studio Code 1.93.1 and "Raspberry Pi Pico" v0.16.2.
python3 is installed via homebrew and can be executed in the terminal. I even provided in the settings of the extension the full path to python3 but no changes. I am not sure where to find any log information by "Raspberry Pi Pico" so I get stuck here.

Interestingly, if I uninstall the extension, remove any cached downloads, an re-install, the first time using "Project from Example" works. Afterwards it is not working anymore.

will-v-pi, 2024-09-25:
Could you try using the fixes in #94 - this should improve the python management by the extension, so may fix this issue? You can download the .vsix file for this PR from https://github.com/raspberrypi/pico-vscode/actions/runs/10998382245?pr=94

mtexplorative, 2024-10-08:
Sorry for taking so long. Thank you very much. It works.

closed 2024-10-08 by mtexplorative