raspberrypi/pico-vscode #89: pico_w compile error

Octopus1633, 2024-09-21:
Extension version: 0.16.2
When the development board is selected as pico_w, the compilation will report an error.
[image]

will-v-pi, 2024-09-23:
Could you confirm what steps you took to create this project? When clicking "New project from example", selecting "blink" example and "Pico W" board, and then creating the project, it compiles successfully for me and only has 81 steps (whereas yours has 175 steps)

Octopus1633, 2024-09-24:
Hello,

I had previously pulled the complete program from [pico-examples](https://github.com/raspberrypi/pico-examples/tree/master) so I did not create a new project. I commented out all other programs in CMakeLists.txt and only compiled blink, and this error occurred.
[image]

I just tried to create a blink project alone, and the compilation still gave an error.
[image]

will-v-pi, 2024-09-24:
It looks like your submodule checkout may have failed when the extension downloaded the SDK - could you try deleting the `~/.pico-sdk/sdk/2.0.0` directory and then re-opening your project, which will re-download the SDK?

Alternatively if you have git installed, you can `cd` into the `~/.pico-sdk/sdk/2.0.0` directory and run `git submodule update init` to checkout the submodules.

Octopus1633, 2024-09-25:
Thank you, this method successfully solved the problem!

closed 2024-09-25 by paulober