raspberrypi/pico-vscode #90: riscv start debugging: libpython3.10.so.1.0: cannot open shared object file: No such file or directory

fhoedemakers, 2024-09-21:
Trying to debug a program compiled for riscv.

Debugger fails with this message:

```
/home/frank/.pico-sdk/toolchain/RISCV_RPI_2_0_0_2/bin/riscv32-unknown-elf-gdb: error while loading shared libraries: libpython3.10.so.1.0: cannot open shared object file: No such file or directory
Error: Unable to start GDB even after 5 seconds or it couldn't even start Make sure you can start gdb from the command-line and run any command like "echo hello".
    If you cannot, it is most likely because "libncurses" or "python" is not installed. Some GDBs require these
GDB session ended unexpectedly. exit-code: 127
GDB could not start as expected. Bad installation or version mismatch. See if you can start gdb from a shell prompt and check its version (Must be >= 9)
```

I created a new project from the blink example

Using Ubuntu 24.04.1 LTS on x64 (WSL)

As far as i can see  python 3.12 is installed on my system, but debugger seems to expect 3.10.

fhoedemakers, 2024-09-23:
Tried it on WIndows. Debugging for ARM works fine, but debugging for riscv fails with:

aiting for gdb server to start...[2024-09-23T15:10:38.073Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\fhoed/.pico-sdk/openocd/0.12.0+dev/openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:\\Users\\fhoed/.pico-sdk/openocd/0.12.0+dev/scripts" -f "c:/Users/fhoed/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl" -f interface/cmsis-dap.cfg -f target/rp2350-riscv.cfg -c "adapter speed 5000"
Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-17-17:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDLiveWatchSetup
Info : Hardware thread awareness created
Info : Hardware thread awareness created
adapter speed: 5000 kHz
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6632891E33DBC33
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x4c013477
Info : [rp2350.dap.core0] datacount=1 progbufsize=2
Info : [rp2350.dap.core0] unavailable.
Error: [rp2350.dap.core0] Examination failed
Warn : target rp2350.dap.core0 examination failed
Info : [rp2350.dap.core1] datacount=1 progbufsize=2
Info : [rp2350.dap.core1] unavailable.
Error: [rp2350.dap.core1] Examination failed
Warn : target rp2350.dap.core1 examination failed
Info : starting gdb server for rp2350.dap.core0 on 50000
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
Error: Target not examined yet
Error executing event gdb-attach on target rp2350.dap.core0:

Error: Target not examined yet
Error: Function FUNC_FLASH_EXIT_XIP not found in RP2xxx ROM.
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected

will-v-pi, 2024-09-23:
You need to make sure your board is running in Risc-V mode before you can attach the debugger - either by using the Run button to run the Risc-V binary, or by running `picotool reboot -c riscv -u` from the integrated terminal to reboot it into Risc-V mode

commit 1ae2b0d4 (2024-09-30): Bump tools and Risc-V toolchain version to v2.0.0-5

Fixes #90
closed 2024-09-30 by will-v-pi