raspberrypi/pico-vscode #95: Can't view (RISC-V) registers in Debug view

NeilBaner, 2024-09-28:
I'm not sure if I'm just looking in the wrong place, but I can't find anywhere where I can see the state of the RISC-V CPU registers of my Pi Pico 2 in the VS Code Debug panel. I can read them with the gdb command `info registers`, but it would be nice to have them in the GUI as well. 

I am running the blink example program in RISC-V mode, modified with an external function implemented in RV assembly. The program runs and behaves as expected. I am debugging with a Raspberry Pi Debug Probe connected to a Raspberry Pi Pico 2. The debugger works as expected after following the instructions in the "Getting Started with Raspberry Pi Pico" guide. 

I was looking for something like the "Registers" panel in Keil MDK: https://www.keil.com/support/man/docs/uv4cl/uv4cl_db_dbg_cpuregs.htm. It's also available in other IDEs like the STM32Cube IDE.

will-v-pi, 2024-09-28:
Yes, the current Cortex Debug extension used for debugging is only designed for Arm Cortex processors, so searches for Arm register names when generating the registers view, not RISC-V. It works fine for everything in RISC-V other than the registers view, which you can still view with `info r`

I’ll leave this open for anyone else who discovers this, so they can see the workaround (to use `info r` manually), but I don’t think we’re planning to fix this as a priority. As far as I’m aware, the Keil MDK and STM32Cube IDEs both only work with Arm processors too, and have no support for RISC-V.

NeilBaner, 2024-09-28:
I see, thanks for the information. I didn't realize the extension uses cortex-debug, but it makes sense that it would take quite a bit of effort to add this feature, then.

closed 2025-02-18 by NeilBaner