raspberrypi/pico-vscode #98: Deleting the build folder.

SaxonRah, 2024-10-07:
In previous versions, I swear I could delete the `build` folder then click `Configure CMake` everything would be fine and I could still compile. No errors.

Now doing the same thing, results in: `ninja: error: loading 'build.ninja': The system cannot find the file specified.`

Running `cmake -G Ninja ..` in the `build` folder after clicking `Configure CMake` in the extension, solves the ninja error.

I had a friend contact me about this same problem, they had accidently deleted the `build` folder, and since I already knew the solution, I was able to solve that for them effortlessly. They apparently searched and struggled with it for a day or two and couldn't figure it out. 

I'm partly posting this issue so people can find a solution if they encounter this, as well ask if this is expected functionality? I don't think I remember running `cmake -G Ninja ..` in the `build` folder before, but I could be wrong as I have to use that command for other toolchains and just can't remember.

will-v-pi, 2024-10-07:
It looks like the notification saying configuration is complete pops up before configuration is actually complete - could you try waiting 10s (or until the `build.ninja` file appears in the build folder) after clicking `Configure CMake` before you click `Compile` and see if that works?

Prototype-Laboratory, 2024-10-14:
Hi, 
I am facing the same issue. The `build.ninja` file did not appear after waiting. The workaround running `cmake -G Ninja ..` in the build folder after clicking `Configure CMake` in the extension, works for me too, but it isn't convenient. I'am under Linux.

Regards

Prototype-Laboratory, 2024-10-14:
Ok, I found the mistake. In one of my `CMakeLists.txt` files in my project folder was the path to the `pico_sdk_import.cmake` file wrong. After I have corrected it, it is working now.

paulober, 2024-10-15:
@SaxonRah Just pushed a commit that should fix this issue. Can you confirm with the [preview build](https://github.com/raspberrypi/pico-vscode/actions/runs/11351963957)?

closed 2024-10-15 by paulober

SaxonRah, 2024-10-16:
> @SaxonRah Just pushed a commit that should fix this issue. Can you confirm with the [preview build](https://github.com/raspberrypi/pico-vscode/actions/runs/11351963957)?

Yes, That does fix the problem. The `build.ninja` file seems to be generated before/at the same time as the popup.

tansiret, 2025-03-15:
This is still an issue btw. I solve it with `cmake -G Ninja ..` thanks to the comments here.

commit c48b2ecc (2024-10-15): Fix #98, CMake configure not awaited