raspberrypi/pico-vscode #101: How to specify SDK as submodule?

Slion, 2024-10-08:
Rather than using an official SDK distribution I'm using the SDK as a submodule.
How to get this working with this extension?

Before it was a simple matter of doing:

```
set(PICO_SDK_PATH "../lib/pico-sdk")
include(pico_sdk_import.cmake)
```

will-v-pi, 2024-10-08:
You should not use a relative path in the CMakeLists.txt file, as that will only work when `cmake` is run from within the build directory - instead use
```cmake
set(PICO_SDK_PATH "${CMAKE_CURRENT_LIST_DIR}/lib/pico-sdk")
```

That should work fine with this extension, provided you put it after the extension setup in CMakeLists.txt (as that also sets `PICO_SDK_PATH`)

Slion, 2024-10-08:
That did it!

closed 2024-10-08 by Slion