raspberrypi/pico-vscode #102: Pico timer gets stuck when Debugged in VS Code Extension

Charry2014, 2024-10-10:
Making my first steps with the Pi Pico WH 2040 board. I am following the instructions in the Pico [Getting Started Guide] to debug a project using the Debug Probe from within the VS Code extension and every example gets stuck in the same place -`cyw43_arch_gpio_put(CYW43_WL_GPIO_LED_PIN, 1);` calls, some layers down the call stack `sleep_until` in time.c, which in turn executes the inline `timer_time_reached`in time.h and there execution gets stuck. It never returns. All calls to sleep_ms do the same thing.

The code I am using is the unmodified [blink example from GitHub](https://github.com/raspberrypi/pico-exa ... wifi/blink). The board is set to pico_w, all other settings are default. This is running on an Intel Mac, macOS Sonoma 14.6.1 using a new Pi Debug Probe straight out of the box, I only updated the firmware following the instructions. 

Putting the 2040 into BOOTSEL mode then simply running the executable using the Run Project (USB) option works fine. Running openocd from the command line also works -
`sudo openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000" -c "program ./blink/build/blink.elf verify reset exit"'

The issue seems to be some strange interaction between OpenOCD in VS Code and the clocks on the target board, not the code or build. What is going wrong?

will-v-pi, 2024-10-10:
This extension currently doesn’t install OpenOCD for Intel Macs, which means that it will use your system OpenOCD. OpenOCD 0.12.0 (the latest version) has this know issue that you’re describing where timers don’t work while debugging - it is fixed upstream but there hasn’t been a new release since.

For Intel Macs using RP2040 you’ll probably be best off installing XPack and installing OpenOCD from there, as they have releases since 0.12.0 which have this bug fixed. Alternatively, if you want RP2350 support then you can compile OpenOCD from source - we have helpful scripts in the pico-sdk-tools repository which we use for Arm Macs, which you could run on your Intel Mac to compile a working copy of OpenOCD.

Charry2014, 2024-10-10:
Thanks for the heads up, much appreciated. There is indeed OpenOCD 0.12.0 installed on the Mac. 

After a bit of fiddling with Node and xpm openocd is updated to 0.12.0+dev-01685-gb9224c0c0-dirty (2024-08-02-22:53) and it is working. Will have to read up on XPack and see how to set up a clean dev environment - the way I did it is horrible, but works.