raspberrypi/pico-vscode #106: Blink example  - compile stops. (correct libc version not found?)

Markjanner, 2024-10-21:
I'm quite new to VS code, and following the online guide to get Blink compiled on a Pi400. Compile stops, seems to be complaining about the library versions? The VScode extension is set to SDK v2.0.0.

I have checked the libc version installed in the path given in the terminal output and it says v2.3.1, which is the latest according version to synaptic. Do I need newer versions from somewhere or is the compiler looking in the wrong place?

[image]

Terminal window:

```
`/blink/build/blink.elf blink.uf2 --family rp2040 --abs-block
/home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool)
/home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool)
/home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool: /lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool)
WARNING: Disassembly is not correct
/home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool)
/home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool)
/home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool: /lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /home/pi400/.pico-sdk/picotool/2.0.0/picotool/picotool)
ninja: build stopped: subcommand failed.`
```

paulober, 2024-10-22:
Thanks for reporting the issue.

I seems like picotool is compiled with a newer GLIBC version as currently included in your OS.
Unfortunately you can't just update this library. 
What is the output of `ldd --version` and `cat /etc/os-release` on your system?

Markjanner, 2024-10-22:
Ah, ok. So I could be looking at needing bookworm, instead of bullseye?

ldd --version:
`ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u11) 2.31`

cat /etc/os-release:
`PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
`

will-v-pi, 2024-10-31:
Yes, the pre-compiled picotool is built on Bookworm, so seems like it doesn't work with Bullseye

closed 2026-11-05 by will-v-pi