raspberrypi/pico-vscode #108: includePath from IntelliSense Configuration not transferred correctly to build.ninja

vansterpc, 2024-10-25:
IAdded include paths in the IntelliSense Configuration editor (which is translated into the c_cpp_properties.json). However, when Configure CMake is run the include paths is not correctly transfered to the build.ninja file. This causes the compilation to fail.

will-v-pi, 2024-10-25:
Yes, that is an IntelliSense configuration, so will only affect the IntelliSense and will not affect compilation at all. To include a file in compilation, you need to add the appropriate lines to CMakeLists.txt - see the CMake documentation for more details. This should also fix the IntelliSense, so you shouldn't need to edit that file in the first place.

closed 2024-10-25 by will-v-pi

vansterpc, 2024-10-25:
Thanks, adding the include_directories() statement in the CMakeLists.txt did fix the issue.