raspberrypi/pico-vscode #109: C++ linker error

Slion, 2024-10-26:
I'm trying to switch from the deprecated Pico Windows setup to this extensions but I'm getting C++ linker errors.

See: https://github.com/raspberrypi/pico-vscode/issues/100#issuecomment-2401789652

will-v-pi, 2024-10-26:
Possibly related to #107 - does changing the compiler in cmake-kits.json to arm-none-eabi-g++.exe help?

Slion, 2024-10-26:
> Possibly related to #107 - does changing the compiler in cmake-kits.json to arm-none-eabi-g++.exe help?

Yes that did it. I changed:

```
"CXX": "${userHome}/.pico-sdk/toolchain/13_2_Rel1/bin/arm-none-eabi-gcc.exe"
```
to
```
"CXX": "${userHome}/.pico-sdk/toolchain/13_2_Rel1/bin/arm-none-eabi-g++.exe"
```

I left the C compiler to:
```
"C": "${userHome}/.pico-sdk/toolchain/13_2_Rel1/bin/arm-none-eabi-gcc.exe"
```

Slion, 2024-10-26:
On #107 it is mentioned that this is not an issue when using the 2.0.0 SDK. I'm currently still using SDK 1.5.1 as a submodule so it could indeed be the case that this is working fine with 2.0.0.

Neril, 2024-10-26:
RE: https://github.com/raspberrypi/pico-vscode/issues/109#issuecomment-2439491563 / @Slion 

> On https://github.com/raspberrypi/pico-vscode/issues/107 it is mentioned that this is not an issue when using the 2.0.0 SDK. I'm currently still using SDK 1.5.1 as a submodule so it could indeed be the case that this is working fine with 2.0.0 somehow.

No, I believe it _is_ an issue on 2.0.0. My follow-up comment on there is just about the second of two additional notes I made about related wrong behavior (the .cmake file name). An example where an attempt to clarify / overload the issue hindered comprehension. Sorry!

will-v-pi, 2024-10-30:
Closing this one as duplicate of #107