raspberrypi/pico-vscode #110: Flash task broken

Slion, 2024-10-26:
Trying to use the Flash task did not work for me.

The following resolved to a non-existing ELF file:
```
\"${command:raspberry-pi-pico.launchTargetPath}\"
```

I had to use that instead:
```
{${command:cmake.launchTargetPath}}
```

I'm guessing the Flash task as it is only works for projects with a single binary output and possibly matching the project name. I have no idea really, just taking a guess.

will-v-pi, 2024-10-26:
You need to use the CMake Tools integration for more complex projects - see the [README](https://github.com/raspberrypi/pico-vscode?tab=readme-ov-file#cmake-tools-extension-integration).

closed 2024-10-26 by will-v-pi

Slion, 2024-10-26:
That did it. I could reverse my changes and it worked.
Any reason why this is not the default?