raspberrypi/pico-vscode #112: "There is no data provider registered that can provide view data."

BruceMardle, 2024-10-26:
I've installed the Raspberry Pi Pico extension under VS Code on a Raspberry Pi 4 running up-to-date Bookworm. When I click on the Pico icon on the left side of the Code windows, it opens a panel that says "There is no data provider registered that can provide view data." How can I fix this?

paulober, 2024-10-28:
This error indicates that the extension failed to activate. Do you see any error/debugging output in the "Help > Developer Tools" console? (or alternatively open a terminal in VS Code then select the output panel and select "Extension Host" in the dropdown on the right)

BruceMardle, 2024-10-28:
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

BruceMardle, 2024-10-28:
but... sizeof(int)=4
[scratches head]

BruceMardle, 2024-10-28:
and...
pi@raspberrypi:~/c $ dpkg --print-architecture
armhf
I'll try installing the OS again.

closed 2024-10-28 by BruceMardle

will-v-pi, 2024-10-28:
Ah yes, I think (although may be wrong) that the 32 bit OS on Pi 4 still has a 64 bit Kernel, but with 32 bit applications - which explains why uname reports 64 bit but dpkg reports 32 bit.