raspberrypi/pico-vscode #114: Ubuntu VSC OpenOCD debug and SWD flash configured as Windows by default

bartola-valves, 2024-10-28:
I have a dual boot machine with Windows and Ubuntu. 
Under Windows (Win 11 pro), I can successfully configure, debug and flash via SWD into a Pico and a Pico2.

Now, under Ubuntu, I can't make it work out of the box despite following all the instructions. I can only get the debug functionality working for the Pico Debug (Cortex Debug with external OpenOCD) only and if I modify the Flash Task.JSON I can get to work the Flash over SWD.

When I look at the tasks.json or the launch.json files when building an out of the box standard Hello Word program, I can see that all the configuration is by default as a Windows machine, not for OpenOCD in a Linux environment

[launch.json](https://github.com/user-attachments/files/17542348/launch.json)
[tasks.json](https://github.com/user-attachments/files/17542351/tasks.json)

Attached are the examples files.

Below is the output of the terminal when executing the SWD flash:

```
 *  Executing task: /home/bartolo/.pico-sdk/openocd/0.12.0+dev/openocd.exe -s /home/bartolo/.pico-sdk/openocd/0.12.0+dev/scripts -f interface/cmsis-dap.cfg -f target/rp2350.cfg -c adapter speed 5000; program "/home/bartolo/Code/BlinkyUbuntu/build/BlinkyUbuntu.elf" verify reset exit 

Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-09-14:16)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter serial' not 'cmsis_dap_serial'
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Hardware thread awareness created
cortex_m reset_config sysresetreq
Error: unable to find a matching CMSIS-DAP device
** OpenOCD init failed **
shutdown command invoked

 *  The terminal process "/home/bartolo/.pico-sdk/openocd/0.12.0+dev/openocd.exe '-s', '/home/bartolo/.pico-sdk/openocd/0.12.0+dev/scripts', '-f', 'interface/cmsis-dap.cfg', '-f', 'target/rp2350.cfg', '-c', 'adapter speed 5000; program "/home/bartolo/Code/BlinkyUbuntu/build/BlinkyUbuntu.elf" verify reset exit'" failed to launch (exit code: 1). 
 *  Terminal will be reused by tasks, press any key to close it. 
 * 
```

Is there any specific way of installing the extension under Ubuntu to get recognised as a Linux environment and get the debug and SWD capability to work out of the box like in Windows?

Thank you

will-v-pi, 2024-10-28:
It looks like the issue is on Ubuntu it can’t find your debug probe - have you set up appropriate udev rules for it to work, as you would for other USB devices on Linux? The extension works out of the box on Rasberry Pi OS, which has the correct udev rules installed, so it does work on Linux environments.

bartola-valves, 2024-11-01:
Yes I have. Would you be able to update me on them to double check that is what I have configured?

 It works perfectly when I run the openocd command on a terminal (it requires sudo though). 
Why does the process invoke "openocd.exe" when it's a linux installation? 
Thank you!

paulober, 2024-11-01:
> Why does the process invoke "openocd.exe" when it's a linux installation?

The .exe doesn't mean its a Windows executable. Actually it's a symlink to the linux binary to have the same path on Windows and other systems. That way your configuration does work cross-platform.

bartola-valves, 2024-11-03:
Thank you for that clarification, it does make sense now! I'm not experienced much in Linux so please bear with me :)

```bash
groups $(whoami)
bartolo : bartolo adm dialout cdrom sudo dip plugdev users lpadmin
```

And here is the other one:

```bash
cat /etc/udev/rules.d/99-picotool.rules
SUBSYSTEM=="usb", \
    ATTRS{idVendor}=="2e8a", \
    ATTRS{idProduct}=="0003", \
    TAG+="uaccess" \
    MODE="660", \
    GROUP="plugdev"
SUBSYSTEM=="usb", \
    ATTRS{idVendor}=="2e8a", \
    ATTRS{idProduct}=="0009", \
    TAG+="uaccess" \
    MODE="660", \
    GROUP="plugdev"
SUBSYSTEM=="usb", \
    ATTRS{idVendor}=="2e8a", \
    ATTRS{idProduct}=="000a", \
    TAG+="uaccess" \
    MODE="660", \
    GROUP="plugdev"
SUBSYSTEM=="usb", \
    ATTRS{idVendor}=="2e8a", \
    ATTRS{idProduct}=="000f", \
    TAG+="uaccess" \
    MODE="660", \
    GROUP="plugdev"
```

much appreciated your help!

will-v-pi, 2024-11-03:
You need to add a rules entry to that file for idProduct=000C, just copy one of those entries and change the idProduct. You’ll need to reboot after changing the file for the changes to take effect.

will-v-pi, 2024-11-03:
Or alternatively copy the openocd rules file to that directory - you can find an example at https://github.com/raspberrypi/openocd/blob/sdk-2.0.0/contrib/60-openocd.rules

bartola-valves, 2024-11-03:
Thank you for the help Will. 
I copied the file above to my rules.d directory, rebooted and worked!

``` bash
ls /etc/udev/rules.d/
10-my-usb.rules        70-snap.canonical-livepatch.rules  70-snap.snapd-desktop-integration.rules  99-cmsis-dap.rules
52-digilent-usb.rules  70-snap.firefox.rules              70-snap.snapd.rules                      99-picotool.rules
60-openocd.rules       70-snap.firmware-updater.rules     70-snap.snap-store.rules
```

You won't believe how frustrating it has been. Since I installed it, I only made it work by running externally OpenOCD with the command. BTW: I couldn't get the Pico2 to work with OpenOCD command despite using the right cfg file for rp2350. 

My only recommendation, ask is:
1. This should be documented in the getting started PDF for Linux installation. When you look at 3.1.2 and Appendix A from the: [getting started PDF](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf), there is simply no mention to this. You will get more people who have limited experience in Linux to get and use the software there. 
2. Error messaging and/or VSC plugin documentation could mention this perhaps?

Thank you for your help!
Ale

commit 6113769f (2024-11-07): Add udev rules to README

Add links to udev rules for OpenOCD and picotool to the README

Fixes #114
closed 2024-11-07 by will-v-pi