raspberrypi/pico-vscode #115: Switch board does not change anything

ObligedTester, 2024-10-29:
Hello,

Recently found out about this vscode extension, seems like a fantastic upgrade to the eco-system!

To my question/issue:
I imported a copy of a project that I used with the SDK manually setup. It asked a couple of questions like SDK version and stuff.

All good, it builds fine and SWD working and so on, however, I wanted to try the feature of switching board to the pico 2.
Unfortunately, nothing seems to change. For example cmake outputs this:
```
[cmake] PICO_SDK_PATH is /home/daniel/.pico-sdk/sdk/2.0.0
[cmake] Defaulting platform (PICO_PLATFORM) to 'rp2040' since not specified.
[cmake] Defaulting target board (PICO_BOARD) to 'pico' since not specified.
```

I did delete my .build to make it as clean as possible. 

Maybe im not using this feature correctly?

will-v-pi, 2024-10-29:
For the switch board feature to work you need a `set(PICO_BOARD…` line in your CMakeLists.txt, which the extension then edits when switching boards - you can see an example of such line by creating a new project. We’re planning to add it to imported projects automatically, see #46

ObligedTester, 2024-10-29:
Ah that is cool! So it is only for imported projects then?

will-v-pi, 2024-10-30:
Yes, this is only an issue for imported projects

To switch build type between release/debug, you need to use the CMake Tools extension integration, and then use the CMake Tools extension to do the switching - see the [README](https://github.com/raspberrypi/pico-vscode?tab=readme-ov-file#cmake-tools-extension-integration) for details

ObligedTester, 2024-10-30:
Thanks a lot. This covers it for me.

Thanks for your help and fantastic work on this extension for the pico!

closed 2024-10-30 by ObligedTester