raspberrypi/pico-vscode #119: Support for custom board files

shalxmva, 2024-11-10:
I have a custom board files on a special folder on my project, and added that folder to PICO_BOARD_HEADER_DIRS.

I looked into the switchBoard.mts file and in askBoard function, only the sdk board folder is used to look for board files.

shalxmva, 2024-11-11:
I created a change, so maybe this will help:

     https://github.com/shalxmva/pico-vscode/tree/develop

Also this change gets the current project_name from CMakeCache.txt file ( my project sets the project name using ${PICO_BOARD}  variable )

The project i am working on is 
   https://github.com/Team-Resurgent/Modxo/tree/sdk2_board_files

will-v-pi, 2024-11-12:
I don't think your fix will work when `PICO_BOARD_HEADER_DIRS` is set to a list, or when it includes relative paths - feel free to raise a PR (you closed #120?) if you want to discuss fixes

> Also this change gets the current project_name from CMakeCache.txt file ( my project sets the project name using ${PICO_BOARD} variable )

I don't think we want to add this, as once you're using CMake variables for the project name you should probably be using the CMake Tools integration as well, which handles all that and more.

will-v-pi, 2024-11-14:
The `main` branch is fine for a PR - as this software is still under development we haven't created a separate `develop` branch yet

will-v-pi, 2025-02-25:
Fixed by #123

closed 2025-02-25 by will-v-pi