raspberrypi/pico-vscode #121: Importing removes custom extensions

trautb, 2024-11-12:
### Expected Behavior
When importing an existing project the neccessary extensions are added to `.vscode/extensions.json`, prior defined "custom" extensions should remain.

### Current Behavior
When importing an existing project `.vscode/extensions.json` gets overridden by the default configuration (independent of prior settings).

### Steps to Reproduce

1. Get pico project
2. Add extension to `.vscode/extensions.json`
3. Import project via Pico extension

### Details
Specifically this `.json`
```json
{
    "recommendations": [
        "marus25.cortex-debug",
        "ms-vscode.cpptools",
        "ms-vscode.cpptools-extension-pack",
        "ms-vscode.vscode-serial-monitor",
        "raspberry-pi.raspberry-pi-pico",
        "trunk.io",
    ]
}
```
gets changed to
```json
{
    "recommendations": [
        "marus25.cortex-debug",
        "ms-vscode.cpptools",
        "ms-vscode.cpptools-extension-pack",
        "ms-vscode.vscode-serial-monitor",
        "raspberry-pi.raspberry-pi-pico",
    ]
}
```

### Context (Environment)

- debian 12 (bookworm), amd64
- VSCodium
```
Version: 1.95.2
Release: 24313
Commit: 7656a0bea17ba6cf301721572e10c9035bb378e6
Date: 2024-11-08T18:21:12.757Z
Electron: 32.2.1
ElectronBuildId: undefined
Chromium: 128.0.6613.186
Node.js: 20.18.0
V8: 12.8.374.38-electron.0
OS: Linux x64 6.1.0-27-amd64
```
- Pico extension v0.17.1

will-v-pi, 2024-11-12:
This probably applies to the other `.vscode/*.json` files too (eg `settings.json`)

closed 2024-11-12 by paulober

paulober, 2024-11-12:
> This probably applies to the other `.vscode/*.json` files too (eg `settings.json`)

Right, I have now added support for merging the recommended extensions. Merging the existing settings can be a bit more complicated as we need to account for each subkey. Probably by switching to the json module to convert between python types and json strings we could get support for keeping settings and such. Although I'm not sure if we really want that for all files like c_cpp_properties.json?

commit c489e3c1 (2024-11-12): Fix #121, Keep ext recommendations on project import + added black formatter