raspberrypi/pico-vscode #124: Integrating Official Raspberry Pi Pico C/C++ SDK

Raart, 2024-11-24:
Python is a nice programming language but it brings some severe limitations.
More low level access is out of discussion and with a dual core MCU this is a real pity.
The solution is already officially ready **Raspberry Pi Pico C/C++ SDK**
[https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf]
The toolchain is partially composed by the same tools used in the Raspberry Pi Pico SDK used here with Python.
I'm to ask the implementation of the C/C++ SDK toolchain in the pico-vscode extension.
Regards

will-v-pi, 2024-11-25:
This extension already uses the Pico C/C++ SDK - it's literally documented as the way to get started with it [here](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)

The MicroPython support was a later addition

closed 2024-11-25 by will-v-pi

Raart, 2024-11-30:
Thank you Will-v-pi, I overlooked this. Happy to now that it's implemented already. Great. A big "thank you" to you and to all the good guys working on this project.